TixiaoShan/LIO-SAM #95: Was the ouster dataset transformed from "LIDAR Coordinate Frame" to the "Sensor Coordinate Frame" already?

BritskNguyen, 2020-09-01:
Dear Author,

According to Ouster's description from their official [website](https://data.ouster.io/downloads/software-user-guide-v1.13.0.pdf?__hstc=34987006.af723c28d603eb06dac2881521b21805.1598671521542.1598944376814.1598967718780.3&__hssc=34987006.1.1598967718780&__hsfp=1124562290) (page 22 - 23) the points are with respect to a "LIDAR Coordinate Frame" {L}, where the positive x direction points from the center towards the socket, and z points upwards as below.

[image]

So if one uses the raw pointcloud data, I assume that the extrinsics should have been set as

```
# Extrinsics (lidar -> IMU)
extrinsicRot: [ 1,  0,  0,
                0, -1,  0,
                0,  0, -1]
extrinsicRPY: [1,  0,  0,
               0,  1,  0,
               0,  0,  1]
```
However the extrinsic params you use seem to be with respect to the "Sensor Coordinate Frame" in the Figure above.

So my question is whether you have done the transform of the pointcloud from the "LIDAR Coordinate Frame" to "Sensor Coordinate Frame" already?

TixiaoShan, 2020-09-01:
I don't think I transform the data. The point cloud data I get is: x - forward, y - left, z - up.

TixiaoShan, 2020-09-01:
> I don't think I transformed the data. The point cloud data I get is: x - forward, y - left, z - up.

The cable of the lidar points to negative x axis.

TixiaoShan, 2020-09-01:
@BritskNguyen 
Actually, I didn't transform the point cloud. I just consider the received cloud from Ouster is in "base_link".

TixiaoShan, 2020-09-01:
It occurs to me that the points in Gen1 and Gen2  lidar might be different. See my replay at https://github.com/TixiaoShan/LIO-SAM/issues/94.

BritskNguyen, 2020-09-06:
> @BritskNguyen
> It occurs to me that the points in Gen1 and Gen2 lidar might be different. See my replay at #94.

After some troubleshooting I think this is indeed the explanation. To recap: gen1 ouster outputs pointcloud in the frame {L} in the figure above. So if we want to use LIO-SAM with gen1 ouster, we should invert the x and y coordinates of each point. It effectively converts the pointcloud to the {S} frame above, which coincides with gen2 ouster and Velodyne.

closed 2020-09-06 by BritskNguyen