TixiaoShan/LIO-SAM #97: gtsam crashed while detecting a loop

pxc2017, 2020-09-02:
hello!as the picture shows.i want to add factor graph and loopclosure in A-loam,odomFactor and gpsFactor and loopFactor are added as you did in lio-sam.But it suddenly crashed while detecting a loop.and the path corrected extremely wrong look for your help thank you a lot

error:  Thrown when a linear system is ill-posed.  The most common cause for this error is having underconstrained variables.  Mathematically, the system is underdetermined.  
[image]
[image]
[image]

TixiaoShan, 2020-09-02:
Have you located which line gives you this error when you test LIO-SAM?

closed 2020-09-11 by stale[bot]

XidianLemon, 2021-06-29:
> Have you located which line gives you this error when you test LIO-SAM? Is it happening at [this line](https://github.com/TixiaoShan/LIO-SAM/blob/7adbdd6e6dec84511763070b9ddd3b25b49a38e4/src/mapOptmization.cpp#L1452) in mapOptimization.cpp?

yes, I also have this problem.
how to fix it?

Alex-Beh, 2022-02-24:
> Have you located which line gives you this error when you test LIO-SAM? Is it happening at [this line](https://github.com/TixiaoShan/LIO-SAM/blob/7adbdd6e6dec84511763070b9ddd3b25b49a38e4/src/mapOptmization.cpp#L1452) in mapOptimization.cpp?

I am facing the same issue and it happen at this [line](https://github.com/TixiaoShan/LIO-SAM/blob/7adbdd6e6dec84511763070b9ddd3b25b49a38e4/src/imuPreintegration.cpp#L301). Below is the report from backward_ros.

```
terminate called after throwing an instance of 'gtsam::IndeterminantLinearSystemException'
  what():  
Indeterminant linear system detected while working near variable
8646911284551352320 (Symbol: x0).

Thrown when a linear system is ill-posed.  The most common cause for this
error is having underconstrained variables.  Mathematically, the system is
underdetermined.  See the GTSAM Doxygen documentation at
http://borg.cc.gatech.edu/ on gtsam::IndeterminantLinearSystemException for
more information.
Stack trace (most recent call last) in thread 267974:
#21   Object "", at 0xffffffffffffffff, in 
#20   Source "../sysdeps/unix/sysv/linux/x86_64/clone.S", line 95, in __clone [0x7fe3999e3292]
#19   Source "/build/glibc-eX1tMB/glibc-2.31/nptl/pthread_create.c", line 477, in start_thread [0x7fe3998a7608]
#18   Object "/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.71.0", at 0x7fe3996de43a, in 
#17   Object "/opt/ros/noetic/lib/libroscpp.so", at 0x7fe39a4562f4, in ros::AsyncSpinnerImpl::threadFunc()
#16   Object "/opt/ros/noetic/lib/libroscpp.so", at 0x7fe39a402f14, in ros::CallbackQueue::callOne(ros::WallDuration)
#15   Object "/opt/ros/noetic/lib/libroscpp.so", at 0x7fe39a402171, in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*)
#14   Object "/opt/ros/noetic/lib/libroscpp.so", at 0x7fe39a454138, in ros::SubscriptionQueue::call()
#13 | Source "/opt/ros/noetic/include/ros/subscription_callback_helper.h", line 144, in operator()
    |   142:   {
    |   143:     Event event(params.event, create_);
    | > 144:     callback_(ParameterAdapter<P>::getParameter(event));
    |   145:   }
      Source "/usr/include/boost/function/function_template.hpp", line 763, in call [0x55ee7a7a7047]
        760:       if (this->empty())
        761:         boost::throw_exception(bad_function_call());
        762: 
      > 763:       return get_vtable()->invoker
        764:                (this->functor BOOST_FUNCTION_COMMA BOOST_FUNCTION_ARGS);
        765:     }
#12 | Source "/usr/include/boost/function/function_template.hpp", line 158, in operator()
    |   156:           else
    |   157:             f = reinterpret_cast<FunctionObj*>(function_obj_ptr.members.obj_ptr);
    | > 158:           BOOST_FUNCTION_RETURN((*f)(BOOST_FUNCTION_ARGS));
    |   159:         }
    |   160:       };
      Source "/usr/include/boost/function/function_template.hpp", line 763, in invoke [0x55ee7a794541]
        760:       if (this->empty())
        761:         boost::throw_exception(bad_function_call());
        762: 
      > 763:       return get_vtable()->invoker
        764:                (this->functor BOOST_FUNCTION_COMMA BOOST_FUNCTION_ARGS);
        765:     }
#11   Source "/home/user/catkin_ws/src/SC-LIO-SAM/SC-LIO-SAM/src/imuPreintegration.cpp", line 385, in odometryHandler [0x55ee7a7a2dcf]
        382:         graphValues.insert(V(key), propState_.v());
        383:         graphValues.insert(B(key), prevBias_);
        384:         // optimize
      > 385:         optimizer.update(graphFactors, graphValues);
        386:         optimizer.update();
        387:         graphFactors.resize(0);
        388:         graphValues.clear();
#10   Object "/usr/local/lib/libgtsam.so.4.0.0", at 0x7fe39a128a7d, in gtsam::ISAM2::update(gtsam::NonlinearFactorGraph const&, gtsam::Values const&, gtsam::FastVector<unsigned long> const&, boost::optional<gtsam::FastMap<unsigned long, int> > const&, boost::optional<gtsam::FastList<unsigned long> > const&, boost::optional<gtsam::FastList<unsigned long> > const&, bool)
#9    Object "/usr/local/lib/libgtsam.so.4.0.0", at 0x7fe39a123748, in gtsam::ISAM2::updateDelta(bool) const
#8    Object "/usr/local/lib/libgtsam.so.4.0.0", at 0x7fe39a119203, in gtsam::ISAM2::Impl::UpdateGaussNewtonDelta(gtsam::FastVector<boost::shared_ptr<gtsam::ISAM2Clique> > const&, gtsam::FastSet<unsigned long> const&, gtsam::VectorValues&, double)
#7    Object "/usr/local/lib/libgtsam.so.4.0.0", at 0x7fe39a11f4fc, in unsigned long gtsam::optimizeWildfireNonRecursive<gtsam::ISAM2Clique>(boost::shared_ptr<gtsam::ISAM2Clique> const&, double, gtsam::FastSet<unsigned long> const&, gtsam::VectorValues&)
#6    Object "/usr/local/lib/libgtsam.so.4.0.0", at 0x7fe39a11ecba, in bool gtsam::internal::optimizeWildfireNode<gtsam::ISAM2Clique>(boost::shared_ptr<gtsam::ISAM2Clique> const&, double, gtsam::FastSet<unsigned long>&, gtsam::FastSet<unsigned long> const&, gtsam::VectorValues&, unsigned long&)
#5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7fe399cc96a8, in __cxa_throw
#4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7fe399cc93f6, in std::terminate()
#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7fe399cc938b, in 
#2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7fe399cbd910, in 
#1    Source "/build/glibc-eX1tMB/glibc-2.31/stdlib/abort.c", line 79, in abort [0x7fe3998e6858]
#0    Source "../sysdeps/unix/sysv/linux/raise.c", line 51, in raise [0x7fe39990718b]
Aborted (Signal sent by tkill() 267963 1000)
```

plane-li, 2022-04-21:
> stale

I also meet the problem. How to solve it??

Alex-Beh, 2022-04-21:
> > stale
> 
> I also meet the problem. How to solve it??

Can you share us your config file and the sensors that you using?

plane-li, 2022-04-21:
> 

Think u !
I use the livox bag,after modify the config file like:
Livox Horizon lidar:
sensor: livox
N_SCAN: 6
Horizon_SCAN: 4000
edgeFeatureMinValidNum: 1
The problem has gone！May be the problem is problem is caused by lidar ponit? I think it may caused by imu?

But I meet an other problem “Not enough features! Only 8 edge and 1394 planar features available.”,when using the livox bag.

However,when i use the **Walking dataset**,It is OK without any error or warning!

plane-li, 2022-04-21:
> > > stale
> > 
> > 
> > I also meet the problem. How to solve it??
> 
> Can you share us your config file and the sensors that you using?

Think u !
I use the livox bag,after modify the config file like:
Livox Horizon lidar:
sensor: livox
N_SCAN: 6
Horizon_SCAN: 4000
edgeFeatureMinValidNum: 1
The problem has gone！May be the problem is problem is caused by lidar ponit? I think it may caused by imu?

But I meet an other problem “Not enough features! Only 8 edge and 1394 planar features available.”,when using the livox bag.

However,when i use the Walking dataset,It is OK without any error or warning!

qqngwzlb, 2023-03-14:
hello，how do you deal this problem？