TixiaoShan/LIO-SAM #100: High velocity with ouster example bag

JimmyFroost, 2020-09-07:
Hi,

First, let me thank you for sharing your great work !

I am new to this slam environment using ROS, so I don't get all parameters right... I have a few ouster OS1-16 rosbags of my own but I want to see how things work with available files here.
I managed to play casualWalk.bag without any problem. Then I made every specified modifications for the Ouster bag but when I play it , I get this : 

[image]

with the message 

> 
 [ WARN] [1599483814.228653537]: Large velocity, reset IMU-preintegration!

I tried to figure it out on the issue pages here but I can't solve this. What should I do ?

TixiaoShan, 2020-09-07:
Since you are using your own bag, you should change the extrinsics according to your own setup.

JimmyFroost, 2020-09-07:
Yes I understand that, but the pictures I show are from the test using the Ouster bags you provided. I would like to see this one work first.

Thanks

TixiaoShan, 2020-09-07:
I just tested the Ouster dataset, it works fine. Some parameters are not set up correctly probably. There is a section in [README](https://github.com/TixiaoShan/LIO-SAM#other-notes) named **Ouster lidar** explains the parameter settings.

JimmyFroost, 2020-09-08:
you're right, I had some forgotten parameters that still needed to be modified, all is well now.
Thanks again for you time.

closed 2020-09-08 by JimmyFroost