TixiaoShan/LIO-SAM #102: confusion about the use of GPS

Gaochao-hit, 2020-09-08:
[image]
Hello, I am confused by the use of GPS. the /navsat node transform the GPS inforamtion into local coordinate. What is the function of node /ekfgps. What's the difference of /imu_raw and /imu_correct,? magnetic_declination?

Gaochao-hit, 2020-09-08:
<joint name="imu_joint" type="fixed">
    <parent link="chassis_link" />
    <child link="imu_link" />
    <origin xyz="0 -0.1 0" rpy="${PI} 0 ${PI}" />
  </joint>
can you also tell me why the relative transformation of imu and chassis is set like this?

TixiaoShan, 2020-09-12:
That parameter is deprecated and not used anymore. The EKF takes in imu_correct, which is just a transformation with an identity matrix.

Gaochao-hit, 2020-09-12:
> That parameter is deprecated and not used anymore. The EKF takes in imu_correct, which is just a transformation with an identity matrix.

Why the magnetic_declination is set zero ? The output of IMU has considered it ?

TixiaoShan, 2020-09-12:
It's parameter decided by the user. You can set it to any number you like.

closed 2020-09-15 by Gaochao-hit