TixiaoShan/LIO-SAM #103: Robosense Compatibility

athtest800, 2020-09-08:
Hello,
Is it compatible with Robosense RS-LIDAR-16 ?
Thank you!

TixiaoShan, 2020-09-12:
It's a mechanical lidar like Velodyne, so it should be compatible with it.

closed 2020-09-18 by athtest800

jpgvl, 2020-11-20:
Hi @athtest800 ,
Could you use the algorithm with an RS-LIDAR-16? I have problems with the "time" and "ring" fields of the pointcloud.

HViktorTsoi, 2020-12-06:
Hi, we develop a tool that can convert Robosense point cloud to Velodyne format seamlessly, without modify the origin algorithm (you may only need to change the parameters like LiDAR beams, vertical resolution, etc.),  we have tested it on LEGO-LOAM, LOAM and LIO-SAM.
You are welcomed to try this
[rs_to_velodyne](https://github.com/HViktorTsoi/rs_to_velodyne)