TixiaoShan/LIO-SAM #106: some confusion about imu orientation, how to set param extrinsicRPY when it's ENU type IMU

Maxinjun, 2020-09-09:
Hi,
according to http://docs.ros.org/independent/api/rep/html/rep-0145.html
the orientation of the world frame may be ENU NED or  aligned  to the power-on orientation of the sensor.
So how to set correct param extrinsicRPY?
And Microstrain 3DM-GX5-25 which the author used is aligned  to the power-on orientation of the sensor, Am I right? @TixiaoShan

Maxinjun, 2020-09-10:
according to  3DM-GX5-25 datasheet 
"The orientation of the sensor frame with respect to the local NED frame"
is there anyone who could explain more clear?

TixiaoShan, 2020-09-12:
Basically, you need to make sure your acceleration, gyro, and RPY align with x-forward, y-left, z-up. The IMU I am using has a different definition in terms of acceleration and RPY. So I have two extrinsics there.

Gaochao-hit, 2020-09-12:
> Hi,
> according to http://docs.ros.org/independent/api/rep/html/rep-0145.html
> the orientation of the world frame may be ENU NED or aligned to the power-on orientation of the sensor.
> So how to set [...]

I think maybe you are right as I find that the quartenion is near identity quartenion when the systen start.

closed 2020-09-20 by stale[bot]