TixiaoShan/LIO-SAM #110: Issue when I am using my own Lidar and IMU data.

jungwoohan72, 2020-09-11:
Thank you for providing such a great algorithm, but now I am having an issue with using my data your algorithm. When I am using my data, RVIZ window shows nothing while it works well with your rosbag data.

I am currently using Velodyne VLP-16 with 9-axis IMU as required. For the IMU sensor, under our experimental setup, it provides two separate IMU information: "/imu/imu", which provides linear acceleration and angular velocity information, and "/imu/pose" which provides orientation information. So, I combined them into one topic, "imu_correct", which publishes sensor_msgs/Imu message type. Of course, I changed "imuTopic" of params.yaml into "imu_correct". 

Do I need extra modification on any file or Lidar data also?

jungwoohan72, 2020-09-11:
Seems like "/tf" topic does not subscribe both "lio_sam_imuPreintegration" and "imu_sam_mapOptimization" nodes...

TixiaoShan, 2020-09-12:
Do you have any output from imageProjection? It should output a point cloud when fed with a raw point cloud and imu data.

jungwoohan72, 2020-09-14:
It publishes to "/lio_sam/deskew/cloud_deskewed" and "lio_sam/deskew/cloud_info" topics, but RVIZ shows nothing.

jungwoohan72, 2020-09-14:
Also, my rosbag publishes its point cloud data to topic named "/vlp_2/velodyne_points", so I changed the "pointCloudTopic" of params.yaml to "/vlp_2/velodyne_points" instead of "points_raw", too.

TixiaoShan, 2020-09-14:
So imageProjection would at lease output "/lio_sam/deskew/cloud_deskewed" and "lio_sam/deskew/cloud_info" with point cloud and IMU as inputs. If you don't have anything from it, it means no point cloud or no IMU is received. You should make sure those two are provided at the same time.

px4Li, 2020-09-16:
@TixiaoShan Thank you for providing such a great SLAM to me. I also have a same problem which the rviz is not able to show point cloud. The error from rviz is "Fixed Frame [map] does not exist".
I also have checked "rostopic echo /lio_sam/deskew/cloud_deskeded" that is no output.
I am using ouster lidar os0-32, and VN-200T-CR.

Here is my part of params.yaml
  # Topics
  pointCloudTopic: "os_cloud_node/points"               # Point cloud data
  imuTopic: "vectornav/IMU"                         # IMU data
  odomTopic: "odometry/imu"                   # IMU pre-preintegration odometry, same frequency as IMU
  gpsTopic: "odometry/gpsz"                   # GPS odometry topic from navsat, see module_navsat.launch file

  # Frames
  lidarFrame: "base_link"
  baselinkFrame: "base_link"
  odometryFrame: "odom"
  mapFrame: "map"

  # GPS Settings
  useImuHeadingInitialization: true           # if using GPS data, set to "true"
  useGpsElevation: false                      # if GPS elevation is bad, set to "false"
  gpsCovThreshold: 2.0                        # m^2, threshold for using GPS data
  poseCovThreshold: 25.0                      # m^2, threshold for using GPS data
  
  # Export settings
  savePCD: false                              # https://github.com/TixiaoShan/LIO-SAM/issues/3
  savePCDDirectory: "~/Downloads/LOAM/"        # in your home folder, starts and ends with "/". Warning: the code deletes "LOAM" folder then recreates it. See "mapOptimization" for implementation

  # Sensor Settings
  N_SCAN: 32                                  # number of lidar channel (i.e., 16, 32, 64, 128)
  Horizon_SCAN: 1024                          # lidar horizontal resolution (Velodyne:1800, Ouster:512,1024,2048)
  timeField: "t"                           # point timestamp field, Velodyne - "time", Ouster - "t"
  downsampleRate: 1                           # default: 1. Downsample your data if too many points. i.e., 16 = 64 / 4, 16 = 16 / 1 
  lidarMinRange: 1.0                          # default: 1.0, minimum lidar range to be used
  lidarMaxRange: 200.0                       # default: 1000.0, maximum lidar range to be used

TixiaoShan, 2020-09-16:
@px4Li 
Your parameters look fine to me. Try to add some ```cout``` to function ```void cloudHandler(const sensor_msgs::PointCloud2ConstPtr& laserCloudMsg)``` and see if this function can reach end of it. If there is no output from ```/lio_sam/deskew/cloud_deskeded```, it means that this function stops somewhere.

jungwoohan72, 2020-09-17:
I found that imageProjection.cpp code stuck on the function bool deskewInfo() in imageProjection.cpp file. Especially on the following part.

        if (imuQueue.empty() || imuQueue.front().header.stamp.toSec() > timeScanCur || imuQueue.back().header.stamp.toSec() < timeScanEnd)
        {
            ROS_DEBUG("Waiting for IMU data ...");
            return false;
        }

I put a cout right before 'return false' line, and it was observed that it keeps returning false. So, the void cloudHandler(const sensor_msgs::PointCloud2ConstPtr& laserCloudMsg) doesn't work properly.

    void cloudHandler(const sensor_msgs::PointCloud2ConstPtr& laserCloudMsg)
    {
        if (!cachePointCloud(laserCloudMsg)) {
            return;
        }

        if (!deskewInfo()) {
            return;
        }

        projectPointCloud();

        cloudExtraction();

        publishClouds();

        resetParameters();
    }

I think there is something wrong with time stamp of IMU queue... Could you give me some help on this issue?

TixiaoShan, 2020-09-17:
```laserCloudIn->points.back().time value returns zero``` doesn't sound right. Have you checked your point cloud data? Make sure they give the right timestamp for each point.

jungwoohan72, 2020-09-18:
The problem was mismatch of incoming IMU and Lidar data. I solved the situation by setting use_sim_time parameter as true.

closed 2020-09-18 by jungwoohan72

px4Li, 2020-09-19:
@jungwoohan72 
Could you please teach me how to set  use_sim_time as true?

jungwoohan72, 2020-09-19:
rosparam set use_sim_time true

Type that before playing the rosbag file.

cg-1728, 2022-06-27:
I have the same problem.
When I set use_sim_time to true, it was still not resolved
Do you know what to do about it?