TixiaoShan/LIO-SAM #111: Ouster OS1 IMU Filter 9-DOF

aliakbarmostafaei, 2020-09-12:
Before describing my issue I'd like to thank you for this great piece of work and contribution, I've been trying to make this work with a bag recorded from an Ouster OS1 and as you already know, OS1's internal IMU is a 6-DOF however by putting it through the [imu_complementary_filter](http://wiki.ros.org/imu_complementary_filter) I managed to simulate a 9-DOF and get the LIO-SAM up and running but I'm still having some issues as indicated in the screenshot attached:

[image]
LIO-SAM is doing some beautiful calligraphy with my dataset! :smiley: 

[image]
A closeup of TF frames.

Here's a summary of what I did:

- Added imu_complementary_filter to the _module_navsat.launch_ and set it's output topic "imu/data" as the default imu topic everywhere in _params.yaml_ .
- Copied OS1's URDF definition from [here](https://github.com/wilselby/diy_driverless_car_ROS/blob/master/rover_2dnav/cartographer_ros/urdf/os1_sensor.urdf) inside config directory and set it as the "robot_description" inside _module_robot_state_publisher.launch_
- Edited the files for Ouster as instructed in the repository's ReadMe.

And here are the issues I'm having so far:

1. The "base_link"  jumps around like a rabbit as you saw in the screenshot!
2. Terminal output is flooded by `Large velocity, reset IMU-preintegration!` and `Not enough features! Only X edge and X planar features available` warnings.
3. Not sure if this is an issue but the "base_link" and the "os1_imu" frames are aligned (NEU) but the "os1_lidar" is facing the opposite direction as indicated in the second screenshot, I guess this is caused by the URDF definitions.

I also uncommented the IMU debug output lines in _imageProjection.cpp_ but I wasn't sure how to output the actual debug info to Terminal.

aliakbarmostafaei, 2020-09-13:
UPDATE: I managed to get the IMU debug output working by remaking the source and here are the first 5 messages as samples:

```
IMU acc: 
x: -0.363919, y: -0.311246, z: 10.3286
IMU gyro: 
x: -0.0153132, y: -0.00359527, z: -0.0234358
IMU roll pitch yaw: 
roll: -0.0301253, pitch: 0.0352036, yaw: -0.000530355

IMU acc: 
x: -0.129287, y: -0.0885855, z: 10.3477
IMU gyro: 
x: -0.00759001, y: -0.00692422, z: -0.0231695
IMU roll pitch yaw: 
roll: -0.0299927, pitch: 0.0349011, yaw: -0.000752406

IMU acc: 
x: -0.445322, y: 0.251391, z: 10.2448
IMU gyro: 
x: -0.0142479, y: -0.00759001, z: -0.0214384
IMU roll pitch yaw: 
roll: -0.029596, pitch: 0.0349054, yaw: -0.000945492

IMU acc: 
x: -0.407014, y: 0.64404, z: 10.082
IMU gyro: 
x: -0.0142479, y: -0.00878843, z: -0.0201069
IMU roll pitch yaw: 
roll: -0.0288106, pitch: 0.0348677, yaw: -0.00111142

IMU acc: 
x: -0.1652, y: 0.550666, z: 10.2879
IMU gyro: 
x: -0.0161121, y: -0.0170442, z: -0.0218379
IMU roll pitch yaw: 
roll: -0.0281544, pitch: 0.0345055, yaw: -0.00129639

```

TixiaoShan, 2020-09-13:
Have you calibrated your IMU and set the params for them? The z acceleration seems very off from 9.8.

Also, please take a look at closed issues that are labeled as Ouster. Some people had success using OS1 with LIO-SAM.

aliakbarmostafaei, 2020-09-13:
@TixiaoShan This may sound noob, but how do I calibrate an OS1's IMU?

And by setting the params, do you mean the information matrix mentioned in the [OS1 User Guide](https://data.ouster.io/downloads/software-user-guide-v1.13.0.pdf) (Page 24) should I insert those elements inside the 2 extrinsic matrices in _params.yaml_?
Right now the two extrinsic matrices inside _params.yaml_ are the default values from the repo, however I did add an [imu_transformer](http://wiki.ros.org/imu_transformer) node to transform the IMU frame to the Lidar frame and I can confirm in my tests it's working.

Through trial and error I managed to fix the base_link jumping around to some extent, not sure if the imu_transformer fixed it or what but I'm getting better results and now the point clouds are only inconsistent in the Z axis! They are drawn stacked and each one beneath the other!

As for the extreme Z acceleration values, I tested the same configuration with a sample [OS1-64 dataset](https://data.ouster.io/sample-data-1.13/OS1-64/index.html) (the OS1-64_city1.bag file to be exact) from Ouster's website, while the Z acceleration in their bag seems reasonable the results are the same! As you can see in the screenshots:

[image]
Ouster's sample dataset
.
.
.
[image]
My dataset

I applied the suggestions from issues #94 and #100 but none produced any working results!

Another odd thing is, I get no messages from any of the` /odometry/gps`,` /odometry/gpsz`,  `/gps/fix` or `/gps/filtered` topics! Only `/odometry/filtered` and` /odometry/navsat` output messages.

Here's my complete src folder, if you wanted to take a look: 
[src.zip](https://github.com/TixiaoShan/LIO-SAM/files/5215135/src.zip)

**Note**: if you wanted to try the Ouster's sample dataset, you have to remove the "/" prefix from the imu frame id. You can clone [this repository](https://github.com/jacoblambert/autoware_bag_tools) and run: 
`rosrun autoware_bag_tools change_frame_id.py -o out.bag -i /path/to/sample-data-1.13_OS1-64_OS1-64_city1.bag -t /os1_cloud_node/imu -f os1_imu`

TixiaoShan, 2020-09-14:
It seems that your IMU data is wrong. You should read the [IMU debug](https://github.com/TixiaoShan/LIO-SAM#prepare-imu-data) in README to debug your data.

aliakbarmostafaei, 2020-09-15:
I changed parameters in params.yaml and Z acceleration is now positive, but the result is the same!
Here's my params configuration:
[params.yaml.zip](https://github.com/TixiaoShan/LIO-SAM/files/5222021/params.yaml.zip)

```
IMU Acceleration: 
X: -2.74854, Y: 0.493206, Z: 8.63589
IMU Gyro: 
X: 0.00692422, Y: 0.0391485, Z: 0.00346211
IMU Orientation: 
Roll: 0.149223, Pitch: 0.127268, Yaw: 0.0472059

IMU Acceleration: 
X: -2.66475, Y: 0.656011, Z: 8.67419
IMU Gyro: 
X: 0.0106526, Y: 0.0443416, Z: 0.00519316
IMU Orientation: 
Roll: 0.14851, Pitch: 0.128972, Yaw: 0.0471154
```
[image]
Screenshot taken from sideways (X axis), as you can see the base_link descending despite the positive Z acceleration!

YushengWHU, 2020-09-15:
> @TixiaoShan This may sound noob, but how do I calibrate an OS1's IMU?
> 
> The sensor was mounted on car, what would be the appropriate values for the 2 extrinsic matrices in _params.yaml_?
> I added an [imu_transformer](http://wiki.ros.org/imu_transformer) node to transform the IMU frame to the Lidar frame and I can confirm in my tests it's working, I guess the IMU and Lidar frames coordination mentioned in the [OS1 User Guide](https://data.ouster.io/downloads/software-user-guide-v1.13.0.pdf) (Page 24) won't be necessary anymore, right?
> 
> Through trial and error I managed to fix the base_link jumping around to some extent, not sure if the imu_transformer fixed it or what but I'm getting better results and now the point clouds are only inconsistent in the Z axis! They are drawn stacked and each one beneath the other!
> 
> As for the extreme Z acceleration values, I tested the same configuration with a sample [OS1-64 dataset](https://data.ouster.io/sample-data-1.13/OS1-64/index.html) (the OS1-64_city1.bag file to be exact) from Ouster's website, and both give negative values and produce similar results! As you can see in the screenshots. Could it be that Ouster also had uncalibrated IMU during their recording?!
> 
> **Ouster's sample dataset:**
> 
> ```
> IMU acc: 
> x: -1.33836, y: -1.30723, z: -8.65025
> IMU gyro: 
> x: 0.00146476, y: 0.00798948, z: -0.00599211
> IMU roll pitch yaw: 
> roll: 0.14283, pitch: -0.159687, yaw: -1.50263
> 
> IMU acc: 
> x: -1.4054, y: -1.05345, z: -8.69095
> IMU gyro: 
> x: 0.00772319, y: 0.00812262, z: -0.00865527
> IMU roll pitch yaw: 
> roll: 0.142914, pitch: -0.159369, yaw: -1.50259
> ```
> 
> [image]
> 
> **My dataset:**
> 
> ```
> IMU acc: 
> x: -0.512359, y: -0.433351, z: -10.3693
> IMU gyro: 
> x: -0.0226369, y: 0.01012, z: 0.0198406
> IMU roll pitch yaw: 
> roll: 0.0256961, pitch: -0.00256328, yaw: -1.65608
> 
> IMU acc: 
> x: -0.344765, y: 0.3304, z: -10.228
> IMU gyro: 
> x: -0.0242348, y: 0.00932106, z: 0.0234358
> IMU roll pitch yaw: 
> roll: 0.0256843, pitch: -0.00196913, yaw: -1.6563
> ```
> 
> [image]
> 
> I applied the suggestions from issues #94 and #100 (plus cleaning and recompiling each time) but none produced any working solution!
> 
> Here's a sample `rostopic delay` output for my dataset:
> 
> ```
> WARNING: may be using simulated time
> average delay: 1597566914.173
> 	min: 1597566914.168s max: 1597566914.178s std dev: 0.00299s window: 20
> average delay: 1597566914.174
> 	min: 1597566914.168s max: 1597566914.182s std dev: 0.00359s window: 40
> average delay: 1597566914.174
> 	min: 1597566914.166s max: 1597566914.184s std dev: 0.00387s window: 60
> average delay: 1597566914.174
> 	min: 1597566914.166s max: 1597566914.184s std dev: 0.00371s window: 80
> average delay: 1597566914.174
> 	min: 1597566914.166s max: 1597566914.184s std dev: 0.00383s window: 100
> ```
> 
> Another odd thing is, I get no messages from any of the ` /odometry/imu`, ` /odometry/gps`,` /odometry/gpsz`, `/gps/fix` or `/gps/filtered` topics! Only `/odometry/filtered` and` /odometry/navsat` output messages.
> 
> Here's my complete src folder, if you wanted to take a look:
> [src.zip](https://github.com/TixiaoShan/LIO-SAM/files/5215135/src.zip)
> 
> **Note**: If you're interested in trying the Ouster's sample dataset, you have to remove the "/" prefix from the imu frame id. You can clone [this repository](https://github.com/jacoblambert/autoware_bag_tools) and run:
> `rosrun autoware_bag_tools change_frame_id.py -o out.bag -i /path/to/sample-data-1.13_OS1-64_OS1-64_city1.bag -t /os1_cloud_node/imu -f os1_imu`

I agree with the author, I can show u the picture when I miset the IMU coordinates
[image]

aliakbarmostafaei, 2020-09-15:
@YushengWHU Okay so how did you solve this? Can you tell me your parameters?
The dataset in the screenshots is a sample from Ouster, you can get it from [here](https://data.ouster.io/sample-data-1.13/OS1-16/index.html), in that page there's also a [sample_config_file.json](https://data.ouster.io/sample-data-1.13/OS1-16/sample_config_file.json) that contains information for Lidar and IMU frame coordination for that dataset as described in the [OS1 User Guide](https://data.ouster.io/downloads/software-user-guide-v1.13.0.pdf) (Page 24). I tried every combination of those values but none fixed this issue!
P.S: I fixed the negative Z acceleration and reported above.

aliakbarmostafaei, 2020-09-15:
I'm guessing the EKF and the NavSat nodes may be the culprit here, since the IMU debug output seems correct (well except for the negative X acceleration which makes the frame move backwards!) but there are sudden jumps and constant decrement in Z axis.

A ROS Graph from my configuration:
[image]

Note: imu_complementary_filter is used to turn the 6-DOF IMU into a 9-DOF stream and imu_transformer is transforming the IMU frame into LiDAR frame, practically doing what the 2 extrinsic parameters are responsible for.

And a tf graph:
[image]

Is this setup correct? Is the "lidar_link" supposed to be a sibling of base_link?

If it's any help this is a sample echo output from /odometry/imu topic, the xyz values in position are dropping rapidly:
```
header: 
  seq: 2572
  stamp: 
    secs: 1573857298
    nsecs: 305729398
  frame_id: "odom"
child_frame_id: "odom_imu"
pose: 
  pose: 
    position: 
      x: -430.109466553
      y: -49.8076934814
      z: -305.0078125
    orientation: 
      x: 0.183668083367
      y: 0.00834548372929
      z: 0.0913653400402
      w: 0.978697482725
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
twist: 
  twist: 
    linear: 
      x: -32.5248591881
      y: -13.0761062392
      z: -29.8175113118
    angular: 
      x: 0.0276143229169
      y: -0.0761537960925
      z: 0.0527808332275
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 2573
  stamp: 
    secs: 1573857298
    nsecs: 315889892
  frame_id: "odom"
child_frame_id: "odom_imu"
pose: 
  pose: 
    position: 
      x: -430.47479248
      y: -49.9008140564
      z: -305.284606934
    orientation: 
      x: 0.183820026747
      y: 0.00791624175774
      z: 0.0915585023687
      w: 0.978654469937
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
twist: 
  twist: 
    linear: 
      x: -32.5444473882
      y: -13.076933372
      z: -29.8265333943
    angular: 
      x: 0.0245516876633
      y: -0.0780180088556
      z: 0.0539792557181
  covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

```

YushengWHU, 2020-09-16:
> @YushengWHU Okay so how did you solve this? Can you tell me your parameters?
> The dataset in the screenshots is a sample from Ouster, you can get it from [here](https://data.ouster.io/sample-data-1.13/OS1-16/index.html), in that page there's also a [sample_config_file.json](https://data.ouster.io/sample-data-1.13/OS1-16/sample_config_file.json) that contains information for Lidar and IMU frame coordination for that dataset as described in the [OS1 User Guide](https://data.ouster.io/downloads/software-user-guide-v1.13.0.pdf) (Page 24). I tried every combination of those values but none fixed this issue!
> P.S: I fixed the negative Z acceleration and reported above, but still it keeps descending!

everything followed what Mr.Shan told me above, and did an extra  calibration of IMU, followed an closed issue of LioSam

closed 2020-09-24 by stale[bot]

HappySamuel, 2020-10-27:
Hi @YushengWHU 

Can you share the ROS package that you use for calibrating IMU? 

Best,
Samuel

hummblebee, 2021-02-18:
I ended up purchasing an INS unit (XSENS MTi-680G IMU+GNSS)  and it seems I have a new problem with synchronizing the Ouster LiDAR and XSENS IMU!
I have contacted support and they said I can only get the GPS data if I integrate them through the Ouster's Interface Box and no IMU data! I tested and I get jerky movement and sudden jumps, and as it is mentioned in the README.md it's because of the un-synced timestamps between LiDAR and IMU messages, is there anyway I can manually sync them by reading the bag file and exporting it to new one?

YushengWHU, 2021-02-19:
> I ended up purchasing an INS unit (XSENS MTi-680G IMU+GNSS) and as it seems I have a new problem with synchronizing the Ouster LiDAR and XSENS IMU!
> I have contacted support and they said I can only get the GPS data if I integrate them through the Ouster's Interface Box and no IMU data! I tested and I get jerky movement and sudden jumps, and as it is mentioned in the README.md it's because of the un-synced timestamps between LiDAR and IMU messages, is there anyway I can manually sync them by reading the bag file and exporting it to new one?

I am using MTI 680G with Ouster OS1-64, 680G is using ros time and you can get the GPS data in ros time timestamp, you dont need to use Ouster interface box, what you need to do is simply PTP sync.
Refererring to the second question, you cant sync them now.

hummblebee, 2021-02-19:
I'm puzzled! So MTi-680G can be connected to the USB port and Ouster OS1 is connected to the Ethernet port, how are they in sync in your setup while not connected to each other? And I believe you're referring to PTP Sync by the Ethernet chipset (which my Ethernet does not support) regarding the Ouster, that still leaves the MTi-680G in the equation, am I right?

Can you please elaborate?

YushengWHU, 2021-02-27:
> I'm puzzled! So MTi-680G can be connected to the USB port and Ouster OS1 is connected to the Ethernet port, how are they in sync in your setup while not connected to each other? And I believe you're [...]

sure， the mti-680g is connected via USB port and the Ouster via Ethernet port, and I am using a DJI manifold 2C for calculation, which support PTP sync. As mti-680G is already synced with ROS time, the only thing is PTP sync Ouster.

hummblebee, 2021-02-28:
Thank you for the reply and sorry for all the bothering, do you think it's also possible with an ordinary Linux laptop? I checked for PTP support on the Ethernet chip and it seems I'm out of luck:
```
Time stamping parameters for enp3s0:
Capabilities:
	software-transmit     (SOF_TIMESTAMPING_TX_SOFTWARE)
	software-receive      (SOF_TIMESTAMPING_RX_SOFTWARE)
	software-system-clock (SOF_TIMESTAMPING_SOFTWARE)
PTP Hardware Clock: none
Hardware Transmit Timestamp Modes: none
Hardware Receive Filter Modes: none
```

Is there any other way to sync these 2 sensors?

Thanks in advance.

YushengWHU, 2021-02-28:
> Thank you for the reply and sorry for all the bothering, do you think it's also possible with an ordinary Linux laptop? I checked for PTP support on the Ethernet chip and it seems I'm out of luck: [...]

As described in the user manual, PTP sync is the simplest and the most recommanded way of sync measure. If your PC cant do a PTP sync, the only way, I suppose, is hardware sync through interface box.

hummblebee, 2021-03-01:
For future visitors, if anyone wants to sync the Ouster with a NIC that does not support hardware timestamps, follow these instructions:

1.  `sudo apt install ptpd`
2. Connect your sensor and wait for the connection to get stable
3. `sudo ptpd -i enp3s0 -M -V`
4. Wait for 140 seconds and you'll 
5. Set the sensor to `TIME_FROM_PTP_1588` timestamp mode and `reinitialize`

You now have PTP sync with your Ouster.

YushengWHU, 2021-03-02:
> For future visitors, if anyone wants to sync the Ouster with a NIC that does not support hardware timestamps, follow these instructions:
>
> 1. `sudo apt install ptpd`
> 2. Connect your sensor and wait [...]

Exactly, I followed this instruction to sync my ouster with ros

Cracks-ll, 2021-12-09:
@hummblebee hello, For imu, what driver are you using?  Is it the official xsens_ros_mti_driver? 
I use it, but the timestamp of the topic "/imu/data" is wrong! I see your timestamp is right. 
Do you know what caused my error?

hummblebee, 2021-12-13:
> @hummblebee hello, For imu, what driver are you using? Is it the official xsens_ros_mti_driver? I use it, but the timestamp of the topic "/imu/data" is wrong! I see your timestamp is right. Do you know what caused my error?

Hi, sorry for the late reply. I wrote a Python script to manually read the timestamps from both devices inside a .bag file and sync them with a given delay (currently set to 8 ms in the script) [here](https://gist.github.com/hummblebee/17a026eae6b2f62bd597560da386120d)'s the gist if you want to experiment.
Let me know if you can get past this issue.