TixiaoShan/LIO-SAM #112: add relTime

LeisureLei, 2020-09-15:
@TixiaoShan I use a rsldiar16 and a MTi-300 to run LIO-SAM. Due to rsldiar16 has no ring and time, I add the calculation of relTime according to LOAM into projectPointCloud() function in imageProjection.cpp .But LIO-SAM becomes worse than before.Is there anything I didn't notice?

TixiaoShan, 2020-09-15:
Make sure the relTime varies between 0 and 0.1 when the lidar rotates at 10Hz. The relative timestamp for the first point received in a scan should be around 0. The last point timestamp should be around 0.1. If the results become worse, I suspect the calculation for the point timestamp is wrong somewhere.

LeisureLei, 2020-09-16:
> Make sure the relTime varies between 0 and 0.1 when the lidar rotates at 10Hz. The relative timestamp for the first point received in a scan should be around 0. The last point timestamp should be [...]

Thanks for your  timely reply!
I checked the timestamp and it is ok. The data was collected while walking. May the low speed cause that?
I also found that there is something wrong in projectPointCloud() function. When I was running the code, I found sometimes ‘i’ of ‘for loop’ suddenly from a number that is not equal zero to zero by print i.

TixiaoShan, 2020-09-16:
```May the low speed cause that?```
I don't think so. Most of my datasets were gathered at walking speed.
``` I found sometimes ‘i’ of ‘for loop’ suddenly from a number that is not equal zero to zero by print i.```
I am not sure what you mean here. Do you mean the poin cloud is empty sometimes?

LeisureLei, 2020-09-17:
> `May the low speed cause that?`
> I don't think so. Most of my datasets were gathered at walking speed.
> ` I found sometimes ‘i’ of ‘for loop’ suddenly from a number that is not equal zero to zero by print i.`
> I am not sure what you mean here. Do you mean the poin cloud is empty sometimes?

Normally,  when the functuion **projectPointCloud()** is called once, **i** in **for loop** will increase from 0 to cloudSize-1 such as:
    
    first time is: 0
    last time is: 0.1

    first time is: 0
    last time is: 0.1
    ... ...

   But sometimes I will get this:
    
    first time is: 0
    first time is: 0
    first time is: 0
    last time is: 0.1

    first time is: 0
    last time is: 0.1
    ... ...
    which means **i** become 0 for several time.
   

    void projectPointCloud()
    {
        float verticalAngle;

        int cloudSize = laserCloudIn->points.size();

        //calculate start angle,end engle
        double startOri = -atan2(laserCloudIn->points[0].y, laserCloudIn->points[0].x);
        double endOri = -atan2(laserCloudIn->points[cloudSize - 1].y,
                          laserCloudIn->points[cloudSize - 1].x) +
                   2 * M_PI;
        if (endOri - startOri > 3 * M_PI)
        {
            endOri -= 2 * M_PI;
        }
        else if (endOri - startOri < M_PI)
        {
            endOri += 2 * M_PI;
        }
        bool halfPassed = false;

        // range image projection
        
        for (int i = 0; i < cloudSize; ++i)
        {
            PointType thisPoint;
            thisPoint.x = laserCloudIn->points[i].x;
            thisPoint.y = laserCloudIn->points[i].y;
            thisPoint.z = laserCloudIn->points[i].z;
            thisPoint.intensity = laserCloudIn->points[i].intensity;

            ////////////////////////////////////////////////////////////////////////////////
            // int rowIdn = laserCloudIn->points[i].ring;                         //
                                                                                  //
            verticalAngle = atan2(thisPoint.z, sqrt(thisPoint.x * thisPoint.x +   //change
                                              thisPoint.y * thisPoint.y)) *       //
                      180 / M_PI;                                                 //
            int rowIdn = (verticalAngle + 15) / 2.0;                              //
            ///////////////////////////////////////////////////////////////////////////////

            if (rowIdn < 0 || rowIdn >= N_SCAN)
                {continue;}

            if (rowIdn % downsampleRate != 0)
                {continue;}

            float horizonAngle = atan2(thisPoint.x, thisPoint.y) * 180 / M_PI;

            static float ang_res_x = 360.0/float(Horizon_SCAN);
            int columnIdn = -round((horizonAngle-90.0)/ang_res_x) + Horizon_SCAN/2;
            if (columnIdn >= Horizon_SCAN)
                columnIdn -= Horizon_SCAN;

            if (columnIdn < 0 || columnIdn >= Horizon_SCAN)
                {continue;}

            float range = pointDistance(thisPoint);
            
            if (range < 1.0)
                {continue;}

            if (rangeMat.at<float>(rowIdn, columnIdn) != FLT_MAX)
                {continue;}
            
            //calculate reltime
            double ori = -atan2(laserCloudIn->points[i].y, laserCloudIn->points[i].x);
            if (!halfPassed)
            { 
                if (ori < startOri - M_PI / 2)
                {
                    ori += 2 * M_PI;
                }
                else if (ori > startOri + M_PI * 3 / 2)
                {
                    ori -= 2 * M_PI;
                }

                if (ori - startOri > M_PI)
                {
                    halfPassed = true;
                }
            }
            else
            {
                ori += 2 * M_PI;
                if (ori < endOri - M_PI * 3 / 2)
                {
                    ori += 2 * M_PI;
                }
                else if (ori > endOri + M_PI / 2)
                {
                    ori -= 2 * M_PI;
                }
            }
            double relTime = (ori - startOri) / (endOri - startOri) * 0.1;
             if(i==0)
             std::cout<<"first time is:"<<relTime<<std::endl;
         
             if(i==cloudSize-1){
                 std::cout<<"last time is:"<<relTime<<std::endl<<std::endl;
             }
            
            thisPoint = deskewPoint(&thisPoint, relTime);  // rslidar
            // thisPoint = deskewPoint(&thisPoint, laserCloudIn->points[i].time); // Velodyne
            // thisPoint = deskewPoint(&thisPoint, (float)laserCloudIn->points[i].t / 1000000000.0); // Ouster
            
            rangeMat.at<float>(rowIdn, columnIdn) = pointDistance(thisPoint);

            int index = columnIdn + rowIdn * Horizon_SCAN;
            fullCloud->points[index] = thisPoint;
        }
    }

TixiaoShan, 2020-09-17:
Is it because the last point at cloudSize-1 is not valid, so the loop is skipped by ```continue```?

LeisureLei, 2020-09-17:
> Is it because the last point at cloudSize-1 is not valid, so the loop is skipped by `continue`?

I think you are right ! I will check carefully !
Besides, I used original edge and surf parameters to test in corridor and there is a serious overlap at both ends of the corridor. Thus I increased  edge and surf parameters, which eliminates the overlap. But there are still altitude errors in corridor. Are there other methods to handle this based lio-sam? I guess lack of ground features cause altitude errors.Thus I tilted the lidar 30 degrees to test and I got worse results.
Also, Can you please tell me how you to calibrate the imu noise? I use [imu_utils](https://github.com/gaowenliang/imu_utils) to calibrate using at least two hours static imu data. And calculated noise parameters also cause worse relust!

TixiaoShan, 2020-09-17:
```But there are still altitude errors in corridor. Are there other methods to handle this based lio-sam? I guess lack of ground features cause altitude errors.Thus I tilted the lidar 30 degrees to test and I got worse results.```
Indoor environments are generally degraded for lidar-based method. It's hard to make sure the features can form fine constraints to the lidar motion. Even if you can cover the ground well, the x - forward motion may not be constrained well. Using it with visual-inertial odometry may solve this problem.
```Can you please tell me how you to calibrate the imu noise?```
I used the exact same package. There are some noise parameters in imuPreintegration.cpp. You can try to tune those manually and see if the results improve.

LeisureLei, 2020-09-17:
> `But there are still altitude errors in corridor. Are there other methods to handle this based lio-sam? I guess lack of ground features cause altitude errors.Thus I tilted the lidar 30 degrees to [...]

I tested with lidar-inertial odometry which solves the problem in corridor!
Thanks for your patient reply！

closed 2020-09-17 by LeisureLei

athtest800, 2020-09-17:
I also plan on using it with rslidar16. Can you provide the working code for this?
Thank you.

LeisureLei, 2020-09-18:
> I also plan on using it with rslidar16. Can you provide the working code for this?
> Thank you.

Just modify the yaml file in condig folder. 

#topic parameter
  **pointCloudTopic**: "rslidar_points"              
  **imuTopic**: "imu/data" 
 
#IMU noise parameters: I use default.
  **imuAccNoise**: 3.9939570888238808e-03
  **imuGyrNoise**: 1.5636343949698187e-03
  **imuAccBiasN**: 6.4356659353532566e-05
  **imuGyrBiasN**: 3.5640318696367613e-05

#Rotate your IMU data to same frame(You may take a while to make sure the  meaning of these parameters)
  **extrinsicRot**: [1, 0, 0,
                 0, 1, 0,
                 0, 0, 1]
  **extrinsicRPY**: [1, 0, 0,
                 0, 1, 0,
                 0, 0, 1]

athtest800, 2020-09-18:
Thanks. How did you overcome the problem of time and ring that the rslidar ros package does not have?

LeisureLei, 2020-09-18:
> Thanks. How did you overcome the problem of time and ring that the rslidar ros package does not have?

Modify the imageProjection.cpp according to lego-loam:

#include "utility.h"
#include "lio_sam/cloud_info.h"

// Velodyne
struct PointXYZIRT
{
    PCL_ADD_POINT4D
    PCL_ADD_INTENSITY;
    uint16_t ring;
    float time;
    EIGEN_MAKE_ALIGNED_OPERATOR_NEW
} EIGEN_ALIGN16;

POINT_CLOUD_REGISTER_POINT_STRUCT (PointXYZIRT,  
    (float, x, x) (float, y, y) (float, z, z) (float, intensity, intensity)
    (uint16_t, ring, ring) (float, time, time)
)

// Ouster
// struct PointXYZIRT {
//     PCL_ADD_POINT4D;
//     float intensity;
//     uint32_t t;
//     uint16_t reflectivity;
//     uint8_t ring;
//     uint16_t noise;
//     uint32_t range;
//     EIGEN_MAKE_ALIGNED_OPERATOR_NEW
// }EIGEN_ALIGN16;

// POINT_CLOUD_REGISTER_POINT_STRUCT(PointXYZIRT,
//     (float, x, x) (float, y, y) (float, z, z) (float, intensity, intensity)
//     (uint32_t, t, t) (uint16_t, reflectivity, reflectivity)
//     (uint8_t, ring, ring) (uint16_t, noise, noise) (uint32_t, range, range)
// )

const int queueLength = 500;

class ImageProjection : public ParamServer
{
private:

    std::mutex imuLock;
    std::mutex odoLock;

    ros::Subscriber subLaserCloud;
    ros::Publisher  pubLaserCloud;
    
    ros::Publisher pubExtractedCloud;
    ros::Publisher pubLaserCloudInfo;

    ros::Subscriber subImu;
    std::deque<sensor_msgs::Imu> imuQueue;

    ros::Subscriber subOdom;
    std::deque<nav_msgs::Odometry> odomQueue;

    std::deque<sensor_msgs::PointCloud2> cloudQueue;
    sensor_msgs::PointCloud2 currentCloudMsg;
    
    double *imuTime = new double[queueLength];
    double *imuRotX = new double[queueLength];
    double *imuRotY = new double[queueLength];
    double *imuRotZ = new double[queueLength];

    int imuPointerCur;
[... 566 lines cut ...]
            cloudInfo.endRingIndex[i] = count -1 - 5;
        }
    }
    
    void publishClouds()
    {
        cloudInfo.header = cloudHeader;
        cloudInfo.cloud_deskewed  = publishCloud(&pubExtractedCloud, extractedCloud, cloudHeader.stamp, lidarFrame);
        pubLaserCloudInfo.publish(cloudInfo);
    }
};

int main(int argc, char** argv)
{
    ros::init(argc, argv, "lio_sam");

    ImageProjection IP;
    
    ROS_INFO("\033[1;32m----> Image Projection Started.\033[0m");

    ros::MultiThreadedSpinner spinner(3);
    spinner.spin();
    
    return 0;
}

athtest800, 2020-09-18:
Thank you very much!

oskarmue, 2022-06-29:
Hey, thank you for your solution! When I try to compile your new code I always get an error, that the variable "timeField" is not declared. 
Furthermore I had to switch "cloudInfo.cloud_deskewed  = publishCloud(&pubExtractedCloud, extractedCloud, cloudHeader.stamp, lidarFrame);" to "cloudInfo.cloud_deskewed  = publishCloud(pubExtractedCloud, extractedCloud, cloudHeader.stamp, lidarFrame);" in order to successfully compile. Is there some error from my side?