TixiaoShan/LIO-SAM #113: Test kitti 00

darkdelting, 2020-09-17:
Thanks for your work ! 
I use your kitti2bag tools to get the 2011-10-03-0027 bag file (sequences 00).Then,I according to Readme.md to modify the params in the params.yaml. Unfortunately,I get a bad result. However,I run the 2011-09-30-0028 (sequences 08) bag file and I got good result. Is that the imu data of sequences00 worse than sequences08.

TixiaoShan, 2020-09-17:
I didn't check the imu data of 00. It's possible the IMU data is not good.

darkdelting, 2020-09-18:
Thanks for your answer.

closed 2020-09-26 by stale[bot]

polbolso, 2021-05-17:
@darkdelting I'm also using KITTI sequence 08 but it seems to be a clear misalignment with the ground truth when evaluating them using evo.
Have you experienced something similar?

[image]