TixiaoShan/LIO-SAM #117: Point cloud data for each trajectory point

jungwoohan72, 2020-09-23:
Thank you for your effort! Could anyone give me some help on following issue?

Seems like 'trajectory.pcd' file stores coordinates of each trajectory point, but is it possible to obtain corresponding point cloud data in addition to the trajectory point?

TixiaoShan, 2020-09-24:
You can export all the point clouds in
```
vector<pcl::PointCloud<PointType>::Ptr> cornerCloudKeyFrames;
vector<pcl::PointCloud<PointType>::Ptr> surfCloudKeyFrames;
```
These clouds are saved in local coordinates. You can use trajectory cloud to transform them into global map frame.

closed 2020-10-02 by stale[bot]

MaxikKot, 2021-03-05:
> 
> 
> You can export all the point clouds in
> 
> ```
> vector<pcl::PointCloud<PointType>::Ptr> cornerCloudKeyFrames;
> vector<pcl::PointCloud<PointType>::Ptr> surfCloudKeyFrames;
> ```
> 
> These clouds are saved in local coordinates. You can use trajectory cloud to transform them into global map frame.

Hi, TixiaoShan! Thanks for the great job.  
Could you please explain how I can use trajectory cloud to transform local coordinateы pointcloud into global map frame?
Thanks a lot in advance!