TixiaoShan/LIO-SAM #118: how to save global coordinate map?

zang09, 2020-09-23:
Can i save cloud map using gps coordinate?

I want to make global map that each cloud points contain gps coordinate, not relative cartesian coordinate.

Of course, i use gps. My own bag files contain '/gps/fix' topic.

Is there any good way to solve this problem??

Thank you.

TixiaoShan, 2020-09-24:
One way to do this is create another keyframe copy similar to
```
vector<pcl::PointCloud<PointType>::Ptr> cornerCloudKeyFrames;
vector<pcl::PointCloud<PointType>::Ptr> surfCloudKeyFrames;
```
Instead of saving points coordinate in the local frame, you can use a new copy to save GPS coordinates for each point.

zang09, 2020-09-24:
I want to use navsat_transform_node in robot localization package to get my utm coordinate.

So I changed params like:
`wait_for_datum: true`
`broadcast_utm_transform: true`
`broadcast_utm_transform_as_parent_frame: true`

But, 'gps/odometry' topic still give me relative xyz, not absolute xyz.

closed 2020-10-02 by stale[bot]

stah-ba, 2025-01-30:
@TixiaoShan could you elaborate a little more on the possibility to add the gps coordinate to each point in the map? I would like to have a map in utm coordinates. I started as you suggested above, but i get the gps position only each few meters and it is inserted to the factor graph for optimizatiion. Can i get the information afterwards for each point from the optimized position also as a global gps position?