TixiaoShan/LIO-SAM #119: ERROR: cannot launch node ...

cristianrubioa, 2020-09-23:
When I compile [roslaunch lio_sam run.launch] the launch terminal gives me:

ERROR: cannot launch node of type [robot_localization/ekf_localization_node]: robot_localization
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/rubio/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
ERROR: cannot launch node of type [robot_localization/navsat_transform_node]: robot_localization
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/rubio/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share

px4Li, 2020-09-24:
@cristianrubioa Have you tried " source /opt/ros/melodic/setup.bash" and "source devel/setup.bash" in lio-sam workspace directory?

cristianrubioa, 2020-09-24:
@px4Li  Thanks. I tried and everything remains the same. I still have the error

px4Li, 2020-09-24:
@cristianrubioa 
maybe you forgot install:

sudo apt-get install -y ros-melodic-robot-localization

cristianrubioa, 2020-09-24:
@px4Li Thanks, solved.

closed 2020-09-24 by cristianrubioa