TixiaoShan/LIO-SAM #120: How to time synch LiDAR with IMU.

px4Li, 2020-09-24:
Hi, @TixiaoShan 
I am using os-32 and vectornav imu for lio-sam as shown in the image below.
[image]

As shown in the figure below, when I run lio-sam, rviz starts to jerking.
[image]

[image]

I have read #94 but I still do not understand to time synch LiDAR with IMU. I did  "ptpd" to do time synchronization of the ouster LiDAR  on my pc. Do you have any suggestions on how to synchronize the IMU?

rosbag: 
https://drive.google.com/file/d/1KxAbhBx3XiXdnXEjLACgTOJZpXvAr9zE/view?usp=sharing

TixiaoShan, 2020-09-24:
It may be caused by vibration. I suggest you calibrate the IMU while the lidar is turned on and static. New IMU parameters may help this problem.

px4Li, 2020-09-27:
@TixiaoShan Thanks for your suggestion. I will try that.
Sorry I have others questions. I would like to ask you. 
1. Here is the image I captured when I run my datasets. The IMU timestamp and LiDAR timestamp are all using same ros time in secs. But do you know why in nsecs they are not same?
[image]

2. The time synchronization does care if the LiDAR and IMU readings are captured at the same time? i.e. They do not have to be synchronized. However, they need to share the same clock, meaning that it is important to know that e.g. the IMU data was taken 4ms after the LiDAR data. For this to be possible, the different clocks must be synchronized.

closed 2020-09-29 by px4Li

IvanVN, 2020-10-16:
Hi @px4Li,

I'm trying to use the same IMU and I'm struggling to get good extrinsic parameters for it. Can you share the configuration you have used, please?

Thank you!

px4Li, 2020-10-20:
Hi @IvanVN 
parameter setting for imu:
[vn100.zip](https://github.com/TixiaoShan/LIO-SAM/files/5406168/vn100.zip)

parameter setting for slam:
[params.zip](https://github.com/TixiaoShan/LIO-SAM/files/5406175/params.zip)

I hope these will give you help.

IvanVN, 2020-10-20:
> Hi @IvanVN
> parameter setting for imu:
> [vn100.zip](https://github.com/TixiaoShan/LIO-SAM/files/5406168/vn100.zip)
> 
> parameter setting for slam:
> [params.zip](https://github.com/TixiaoShan/LIO-SAM/files/5406175/params.zip)
> 
> I hope these will give you help.

Thank you very much! My IMU is in a different position, but it has helped me a lot to find where my error was. :)

JJuhyeon-Kim, 2021-03-11:
> @TixiaoShan Thanks for your suggestion. I will try that.
> Sorry I have other questions. I would like to ask you.
> 
> 1. Here is the image I captured when I run my datasets. The IMU timestamp and LiDAR timestamp are all using same ros time in secs. But do you know why in nsecs they are not same?
>    [image]
> 2. The time synchronization does care if the LiDAR and IMU readings are captured at the same time? i.e. They do not have to be synchronized. However, they need to share the same clock, meaning that it is important to know that e.g. the IMU data was taken 4ms after the LiDAR data. For this to be possible, the different clocks must be synchronized.

Hi, @px4Li !

You gave me the idea of using ptpd, I cound sync the timestamp of LiDAR. 
I'm really grateful to you.

I have a question for you, are you still going through this problem?
I'm using OS1-128 (GEN2), and I'm having the same problem as issue207.
https://github.com/TixiaoShan/LIO-SAM/issues/207
I solved the timestamp problem, but the mapping is unstable like you.
Did you solve this problem?