TixiaoShan/LIO-SAM #121: Testing on Ford Autonomous Vehicle Dataset

Pallav1299, 2020-09-25:
I have been trying to test LIO-SAM on the Ford AV Dataset ([LINK](https://avdata.ford.com/home/default.aspx)). I used the extrinsics information available in calibration files. Alike KITTI, IMU intrinsics are unavailable. I am facing the following issues:

1. No data in **/odometry/gps" topic**, although **sensor_msgs/NavSatFix** and  **sensor_msgs/Imu** data is available.
2. The map is being stitched vertically inverted. I suspect it to be a difference in **LIDAR -> IMU** extrinsics. Am I right?

Here is the sensor setup.
[image]

closed 2020-10-04 by stale[bot]

HaisenbergPeng, 2021-03-23:
Hi @Pallav1299 , I am trying to use Ford AV dataset as well. Here is what I did. Following the ".urdf" file provided by the dataset, I get the imu frame origin expressed in lidar frame [-1.2115, -0.4846, 0.8182] as extrensicTrans; I get the rotation matrix that rotates imu coordiates to align with lidar coordinates[-0.0007,-0.4265,0.9045,  0.9999, -0.0120, -0.0049,  -0.0129, -0.9044,-0.4265], as extrensicRot and extrensicRPY. But still the error "MSG to TF: Quaternion Not Properly Normalized" comes out. It must be the wrong transformation.

Pallav1299, 2021-03-23:
@HaisenbergPeng, I was able to get it up and running.

> But still the error "MSG to TF: Quaternion Not Properly Normalized" comes out. It must be the wrong transformation.

As I remember, Ford setup consists of a 6-axis IMU. There is no global orientation data from IMU which leads to some conflicts with the quaternion magnitude check. That's why the error. 
I was able to get it working, but the map orientation is not correct with respect to global reference.

HaisenbergPeng, 2021-03-24:
@Pallav1299 May I ask what your parameter setting for extrensicTrans and extrensicRot is? 
Yes indeed the Ford setup uses 6-axis IMU. However, what bothers me is that I don't see where the global orientation of IMU data is required, except for the function "failureDetection", since the author uses integration module built in gtsam to get the orientation.
As for the map orientaion, I haven't gotten that far yet.

HaisenbergPeng, 2021-04-01:
@Pallav1299 Thanks for your kindly reply. I just figured out how you got those parameters. But when I tried to test LIO-SAM, I still couldn't get it working. Could you explain in brief what major changes you made to the source code? for example, how did you cope with the '6-axis IMU' problem? Greatly appreciated!

BurhanMuhyiddin, 2022-08-28:
Hi all. I also try to test LIO-SAM with ford sample dataset. I have found extrinsicROT and Translation as below:

```
extrinsicTrans: [-0.345, -1.224, -1.000]
extrinsicRot: [-5.65756149e-03,  9.99983996e-01,  -2.40066258e-05, 9.99947986e-01,  5.65756149e-03, -8.48634223e-03, -8.48634223e-03, -2.40066258e-05, -9.99963990e-01]
extrinsicRPY: [-5.65756149e-03,  9.99983996e-01,  -2.40066258e-05, 9.99947986e-01,  5.65756149e-03, -8.48634223e-03, -8.48634223e-03, -2.40066258e-05, -9.99963990e-01]
```
I calculated them directly using TF static topic from rosbag files. Namely, relative transformation between frames lidar_blue and imu. However, the result is catastrophic. There are zig-zag behaviors and the map is not correct. Do you know what should I do else? Also, I couldn't understand the difference of using 9axis IMU and 6axis IMU, if the datasets only providing rosbag files which contain sensor_msgs/Imu. The structure of the IMU message is the same for both 6axis and 9axis IMU? So, what is the matter? I would like to note that even I use ford dataset, I don't get the error of quaternion magnitude less than 0.1.Thank you very much.