TixiaoShan/LIO-SAM #127: The "base_link" shaking when I mapping

rensixu, 2020-09-30:
Hi, just as the title descript. I found the "base_link" shaking when I use my own dataset mapping in rviz, that make the cloudinput looks not good matching to the map. But the map could be built correctly and the  trajectory is right too. I do not know why.

closed 2020-10-09 by stale[bot]

jjbecomespheh, 2021-01-29:
Hi @rensixu @TixiaoShan , I have the same problem too. It was mentioned in Readme that base_link jumping up and down is because of IMU extrinsics ie. negative gravity value but I have checked, my gravity value is positive.

CanCanZeng, 2021-09-16:
Hi, I meet the same problem, have you solved it out? @jjbecomespheh  @rensixu 
I find that the preintegrated orientation is very different from the optimized, but I just save the imu data used by preintegrator and then integrate is myself, it is very similar with the optimized pose. So, I guess the bais estimated is not right since my manual integration omits bais.

maxkremmel, 2023-05-15:
I'm having the same issue. Is there a solution to it?