TixiaoShan/LIO-SAM #128: How to connect the IMU to a ROS system for consistent stream data logging?

tsurudyne, 2020-10-02:
Hi @TixiaoShan , thank you for your excellent work and beautiful source code.

When I tried mapping with your dataset(casual_walk.bag, park.bag, etc..), I was able to generate clear and detailed maps.
However, when I tried to verify the mapping using own data, I had a problem.
The lidar I used is VLP16 and the IMU is 3DM-GX5-25.
The mapping seems to be successful, but it looks like the point cloud of the object is noisier than when using your dataset.

I'm guessing this is a problem with the IMU data.
According to the 3DM-GX5-25 [manual](https://www.microstrain.com/sites/default/files/3dm-gx5-25_dcp_manual_8500-0065.pdf)(p.28), data streaming is inconsistent when using USB.
However, I connect the IMU and PC via USB and use ros_mscl to set imu_data_rate to 500Hz for logging.
Therefore, when I checked my rosbag with rqt_bag, the data was inconsistent and multiple data were gathered at about the same moment.

I think you've solved this problem, please tell me how you connected the IMU to the ROS system.

TixiaoShan, 2020-10-10:
Hi @tsurudyne 
I used USB to connect the IMU. Actually, I may have the same problem, but I never checked. Here is my IMU driver. If you are interested, you can try it out.
[microstrain_mips.zip](https://github.com/TixiaoShan/LIO-SAM/files/5359177/microstrain_mips.zip)

tsurudyne, 2020-10-12:
> Hi @tsurudyne
> I used USB to connect the IMU. Actually, I may have the same problem, but I never checked. Here is my IMU driver. If you are interested, you can try it out.
> [microstrain_mips.zip](https://github.com/TixiaoShan/LIO-SAM/files/5359177/microstrain_mips.zip)

Thank you for your reply!
I used the driver you provided and it worked well !
My imu data is now similar to your data.

There seems to be a problem with latest [ROS-MSCL](https://github.com/LORD-MicroStrain/ROS-MSCL) package(or MSCL driver).
https://github.com/LORD-MicroStrain/MSCL/issues/53
https://github.com/ros-drivers/microstrain_mips/issues/32

Anyway, I'll try to  experiment with mapping tomorrow.

I am looking forward to your research presentation at IROS.
Thank you.

closed 2020-10-12 by tsurudyne