TixiaoShan/LIO-SAM #129: lidar horizontal resolution

ganindu7, 2020-10-07:
Hi can you please explain 
```
Horizon_SCAN: 1800                          # lidar horizontal resolution (Velodyne:1800, Ouster:512,1024,2048)
```
I know that VLP 16 outputs 384 packets per scan, @0.2 Azimuth res at 600 RPM. I don't get how 1800 fits in. my code works but this is bugging me. I've used 3 lidar manufactures (Robosense, RFANS and LS) and all seem to work without this value being changed. (robosense and LS I've personally tested)

closed 2020-10-08 by ganindu7

iportolesrb, 2020-11-20:
Hi @ganindu7,
How did you test it with the robosense lidar? The driver does not publish the "time" and "ring" fields in the pointcloud, does it?

ganindu7, 2020-11-20:
HI @iportolesrb I wrote a new driver. Basically you just have to augment the point type. Have a look at point types in PCL and the datasheet for the device. The one I self-solved is the only problem I had.
Cheers,
Ganindu.

DamienSalle, 2020-12-05:
@ganindu7, I'd also want to use a rs-lidar16 with lio-sam. Could you please share your new driver publishing the time and ring fields?
That would help a lot! 
Thanks
Damien