TixiaoShan/LIO-SAM #130: Initial Pose

mysterybc, 2020-10-12:
Is Lio-sam sensitive to the initial pose? 
Should pitch&roll be zero  when i start the algorithm?
I meet some problem when I tested the method with my car, and I thought it might be casued by unflatness of the floor.
Thank you in advance for the help!

TixiaoShan, 2020-10-12:
It's not sensitive because IMU gives roll and pitch for initialization. The roll and pitch don't need to be zero. You can even initialize the system at various angles, say 45-degree slope.

mysterybc, 2020-10-12:
Thanks! 
And one more thing, is the Euler angle only used for initialization? Since my IMU only provide acceleration and angular velocity, the Euler angle is calculated by myself, so it might not be accurate, will it cause lio-sam diverge? (and it did fail after a short distance running in our corridor)

TixiaoShan, 2020-10-12:
Euler angle is used for both initialization and pose correction. If it's not accurate, the mapping results won't be accurate. You can try to set the value at [here](https://github.com/TixiaoShan/LIO-SAM/blob/7f3fc4b53587401653c1a13d1d6c44eeda7ec6f9/src/mapOptmization.cpp#L1262) and [here](https://github.com/TixiaoShan/LIO-SAM/blob/7f3fc4b53587401653c1a13d1d6c44eeda7ec6f9/src/mapOptmization.cpp#L1609) to zero to not use Euler angles for correction.

mysterybc, 2020-10-16:
I've tested the algorithm after set those params to zero, the result is still unsatisfied. I think I have to make sure the IMU measurement is correct first, thanks for your help !

cqrtxwd, 2020-10-21:
hi, my imu don't provide imu pose, and my imu corrdinate is z-axis inverse, is there any chance I can run the system properly with little modify?

closed 2020-11-06 by stale[bot]