TixiaoShan/LIO-SAM #133: The error result of my own dataset (drift)

JJuhyeon-Kim, 2020-10-19:
Hi, first of all, thank you for revealing a really good study.
Your research has become a great example to me.

I'm trying a lot of experiments with your algorithm.
However, there are times when I fail, so I want to ask you why.
I made all the conditions the same as your research.

**1. Pedestrian overpasses**
(2020-10-08-16-34-44.bag)
My guess about the failure of the first experiment is that the model was confused because it took data for a long time in one place. But I fail every time I do an experiment on pedestrian overpass. I don't know why.
[image]
[image]

**2. Roof Structure**
(2020-10-14-11-40-55.bag)
My guess is that the second experiment failed because I got the data from the roof and failed to match the features. Or maybe it's because the distance between nearby structures and other structures is huge.
[image]

I uploaded a link to my file. I would really appreciate it if you answer me.
https://drive.google.com/drive/folders/1lY_JQEvojOPYO8m5imEqBzqxve_Rq9zU?usp=sharing

TixiaoShan, 2020-10-19:
Thanks for sharing the data. Can you share your extrinsics?

JJuhyeon-Kim, 2020-10-20:
I think you must be very busy, but thank you so much for replying so quickly.
Here is my params.yaml file
I didn't change anything from you file except topic name. 

https://drive.google.com/drive/folders/1F_kMXPAdxfhmymauiMEeppUkxK9Gc26d?usp=sharing

TixiaoShan, 2020-10-21:
It looks like the lidar undergoes very aggressive motion. LIO-SAM may not be able to handle that. Have you tried other methods?

JJuhyeon-Kim, 2020-10-21:
Thank you for your answer.
The results came out well in other experiments.
As you said, I will acquire data from the rooftop and under the overpass in a stable environment.
Thank you very much.

closed 2020-10-21 by JJuhyeon-Kim