TixiaoShan/LIO-SAM #134: When I use GPS,the map direction is different from when GPS is not used

qian5683, 2020-10-21:
when I use GPS, it makes the map rotate, and the direction isn't the IMU anymore.

Pallav1299, 2020-10-22:
Please verify the following:
1. 9-axis IMU is being used.
2. **useImuHeadingInitialization** is set to **true** in **params.yaml**

cklbywq, 2020-10-28:
> when I use GPS, it makes the map rotate, and the direction isn't the IMU anymore.

Hi, can you tell me how to add GPS data in the project? I tried but no output in topic /odometry/gps and /odometry/navsat. Or can you show me your params.yaml and module_navsat.launch?

closed 2020-11-13 by stale[bot]

FishInWave, 2021-11-20:
> > when I use GPS, it makes the map rotate, and the direction isn't the IMU anymore.
> 
> Hi, can you tell me how to add GPS data in the project? I tried but no output in topic /odometry/gps and /odometry/navsat. Or can you show me your params.yaml and module_navsat.launch?

I meet the same question. Did you finally solve it？

Pallav1299, 2021-11-21:
@FishInWave, I think your issue is already addressed. Please go through the known issues in README.