TixiaoShan/LIO-SAM #137: Some problems about devices and coordinate settings

Aragony, 2020-10-21:
Hi @TixiaoShan, 
Thanks for sharing your excellent work. 
I met some problems when I try to run the code with my devices, a VLP-16 and 3DM-GX5-45. The odometry drift a lot even when the device is static at the beginning.
I thought this may be caused by the incorrect extrinsic in the params.yaml, and I tried to modify the extrinsic according to the readme and your replies in issues. But the code still doesn't work. 
I set the LiDAR coordinate frame as x - right, y - forward, z - upward, and IMU coordinate frame as x-left, y-forward, z-downward. Under this settings, I set the extrinsicRot and extrinsicRPY as: 
-1 0 0
0 1 0
0 0 -1 
Is it right?
My questions are:
In my opinion, if I clearly define the LiDAR and IMU frame and provide the extrinsic between them, the code should work. I cannot understand why it is necessary to follows the ROS REP-105 convention. In our settings, the device can determine the DOWN direction, but nobody defines FORWARD and LEFT if we don't mount the device on a vehicle. right? 
According to the readme, your IMU 3DM-GX5-25 has different coordinate frames for acceleration and attitude. I am using a simular IMU, 3DM-GX5-45, and I checked their manuals and found that they all output the attitude in local NED frame. By simple test, I found that the local NED frame coincide with the initial sensor frame. (I dont use the magnetometer in the IMU setting. ) So I think the extrinsicRPY should be the same as the extrinsicRot . Could you please explain the reason why you use different rotation matrix? It would be great help if you can provide your IMU settings. 
According to the readme and the coordinate definition figures you provide in readme, I think the extrinsicRot and extrinsicRPY should be the transformation of the LIDAR data from LiDAR frame to IMU frame. But I also noticed in an issue-talk you said they are the TF from IMU frame to LiDAR frame, which makes me confused. Would you please provide a clear definition?
I am using almost the same devices as your configuration, so I believe it should be easy to make the code works, but without luck by far. It would be great appreciation if you can give me other clues.
Thank you in advance.
Wang Shuo

TixiaoShan, 2020-10-21:
The IMU driver I am using can be found here -> https://github.com/TixiaoShan/LIO-SAM/issues/128 Maybe this driver will help?
I also have a 3DM-GX5-45. Some bags, including Campus Large and Small shared in this repo, were gathered using this IMU.

Aragony, 2020-10-22:
> The IMU driver I am using can be found here -> #128 Maybe this driver will help?
> I also have a 3DM-GX5-45. Some bags, including Campus Large and Small shared in this repo, were gathered using this IMU.

Thank you for your reply. I notice that the imu topic is "/imu_correct" in the Campus Small bag. Is there any transformation between imu_raw and imu_correct?  In other words,
> The "extrinsicRot" and "extrinsicRPY" in "config/params.yaml" needs to be set as identity matrices.
it means the raw data between liDAR and IMU or correct data ?
Thank you in advance.
Wang Shuo

TixiaoShan, 2020-10-22:
imu_raw is just the original /imu/data. imu_correct is the transformed data in the imu driver. 

It's the correct data.

vislero, 2020-10-26:
HI @TixiaoShan 
what is the 'transformed' mean? 

is it just coordinate transform between frames by matrix multiplication? 

or another filter ( e.g kalman filter) is applied to raw data ?

Aragony, 2020-11-05:
Thank you for your reply. I used your IMU driver and I got a wonderful result. But I'm confused in microstrain_3dm.cpp line 3714 - line 3730
```cpp
              if(frame_based_enu_)
              {
                // Create a rotation from NED -> ENU
                tf2::Quaternion q_rotate;
                q_rotate.setRPY(M_PI,0.0,M_PI/2);
                // Apply the NED to ENU rotation such that the coordinate frame matches
                q = q_rotate*q;
                quat_msg = tf2::toMsg(q);
              }
              else
              {
                // put into ENU - swap X/Y, invert Z
                quat_msg.x = q[1];
                quat_msg.y = q[0];
                quat_msg.z = -1.0*q[2];
                quat_msg.w = q[3];
              }
```
I think what you do in if and else is same. Am I think right? And I can't understand what your code means in else part. Could you please tell me it?
Thank you in advance!

closed 2020-11-21 by stale[bot]