TixiaoShan/LIO-SAM #138: Unable to use GPS data

cklbywq, 2020-10-23:
Hello, I have an issue that when I want to add GPS data into the program,there are datas in topic /imu/data and /gps/fix but no output in topic /odometry/navsat and /odometry/gps. Could you please tell me why? Below is my config file.

[params.txt](https://github.com/TixiaoShan/LIO-SAM/files/5428016/params.txt)

GilenTell, 2020-10-27:
Hi, I have the same problem.
I have integrated a ublox ZED-F9P RTK GPS. The ROS driver publishes /gps/fix topic with frequency of 4Hz.
The other topics are /imu and /velodyne_points.
I have modified the module_navsat.launch replacing /odometry/navsat with /odometry/gps and remap /imu/data to /imu, but same result. There isn't any output in /odometry/gps.
Any suggestion?
Thanks!

closed 2020-11-13 by stale[bot]

HMellor, 2020-12-18:
I'm having the same issue. (Also using the F9P, but I don't think that's the issue because `rostopic echo gps/fix` is showing that the receiver is publishing correctly)

Pallav1299, 2020-12-18:
Please check if there is transform between the link in the message "frame_id" and robot_frame_id. For e.g. in my case I was missing the transform between "base_link" and "imu_link". This is a requirement of robot localization. 
Link: http://docs.ros.org/en/melodic/api/robot_localization/html/preparing_sensor_data.html

[image]

GilenTell, 2020-12-22:
Yes, in my case, there was missing a Transform because my bag topics need it. Now the sensor fusion and odometry is working.
This is what I have introduced in run.launch.
```
<!--  Transforms  -->
<node pkg="tf" type="static_transform_publisher"
    name="gps_transform"
    args="0 0 0 0 0 0 base_link gps 10" />

<node pkg="tf" type="static_transform_publisher"
        name="imu_transform"
        args="0 0 0 0 0 0 base_link base_imu_link 10" />
```
And I have changed thesetwo parameters becaus my SEN-14001 is aligned with North.
```
magnetic_declination_radians: 0.0061
yaw_offset: 1.5707963
```
But I have another issue. Now the mapping is drifting. And a warning is reported:
`Warning: Invalid argument passed to canTransform argument target_frame in tf2 frame_ids cannot be empty at line 126 in /tmp/binarydeb/ros-melodic-tf2-0.6.5/src/buffer_core.cpp`

I don't know if the reason is that my IMU isn't so much precise, or that the RTK GPS sometimes loss the quality of position and varies some meters.
Any advice?

Thank you!

Zahra-Farajzadeh, 2024-10-17:
> Yes, in my case, there was missing a Transform because my bag topics need it. Now the sensor fusion and odometry is working. This is what I have introduced in run.launch.
> 
> ```
> 
> <node pkg="tf" type="static_transform_publisher"
>     name="gps_transform"
>     args="0 0 0 0 0 0 base_link gps 10" />
> 
> <node pkg="tf" type="static_transform_publisher"
>         name="imu_transform"
>         args="0 0 0 0 0 0 base_link base_imu_link 10" />
> ```
> 
> And I have changed these two parameters because my SEN-14001 is aligned with North.
> 
> ```
> magnetic_declination_radians: 0.0061
> yaw_offset: 1.5707963
> ```
> 
> But I have another issue. Now the mapping is drifting. And a warning is reported sometimes: `Warning: Invalid argument passed to canTransform argument target_frame in tf2 frame_ids cannot be empty at line 126 in /tmp/binarydeb/ros-melodic-tf2-0.6.5/src/buffer_core.cpp`
> 
> I don't know if the reason is that my IMU isn't so much precise, or that the RTK GPS sometimes loss the quality of position and varies some meters. Any advice?
> 
> Thank you!

did you solve this issue?
`Warning: Invalid argument passed to canTransform argument target_frame in tf2 frame_ids cannot be empty at line 126 in /tmp/binarydeb/ros-melodic-tf2-0.6.5/src/buffer_core.cpp`

PR #176 (2020-12-23): Added info about GPS odometry unavailability issue
Adding information for solving the GPS odometry unavailability issue to README.md.
Fixes #138