TixiaoShan/LIO-SAM #142: GTSAM was built against a different version of Eigen.

Danis199, 2020-10-27:
Hi everyone! Maybe someone faced such a problem when building a package?
[image]

cklbywq, 2020-10-28:
Hi, you can add "set(GTSAM_USE_SYSTEM_EIGEN ON)" before "if(GTSAM_USE_SYSTEM_EIGEN)  find_package(Eigen3 REQUIRED)" in CMakeLists.txt of gtsam, and then re-compile gtsam again.

Danis199, 2020-10-28:
Thank you! It's work.

closed 2020-10-28 by Danis199

AndreiBarsan, 2021-02-20:
This helped me as well. By the way, for the sake of simplicity, you can achieve the same effect by simply passing an additional flag to `cmake`: `-DGTSAM_USE_SYSTEM_EIGEN=ON`

congphase, 2021-09-01:
I modified `CMakeLists.txt` as suggested. But it does not even have a change I think.

[image]

[image]

hellojarmon, 2022-09-29:
> I modified `CMakeLists.txt` as suggested. But it does not even have a change I think.
> 
> [image]
> 
> [image]
> 
> Do you have any idea?

Did you solve it ?

congphase, 2022-10-29:
@hellojarmon Have you tried running the ["Using Docker"](https://github.com/TixiaoShan/LIO-SAM#using-docker) in the README?

Back then (Sept 2021), I solved it by creating a Docker image to resolve all the painful conflicts. But it was in Sept 2021, the repo did not support Docker stuff. Now they have one, you should use them, it should work.