TixiaoShan/LIO-SAM #147: ERROR: Point cloud is not in dense format, please remove NaN points first!

zcg3648806, 2020-11-03:
When I run the KITTI raw data, I encountered the following errors：

[ERROR] [1604370963.222598901]: Point cloud is not in dense format, please remove NaN points first!
[lio_sam_imageProjection-3] process has finished cleanly
log file: /home/zcg/.ros/log/4231a780-1d7d-11eb-8354-98eecb890c9c/lio_sam_imageProjection-3*.log
[lio_sam_imageProjection-3] restarting process
process[lio_sam_imageProjection-3]: started with pid [11828]
[ INFO] [1604370963.481902023]: ----> Image Projection Started.
[ERROR] [1604370963.938057740]: Point cloud is not in dense format, please remove NaN points first!

How can I solve this problem？Thanks！

VRER1997, 2020-11-03:
Do not use kitti2bag installed by pip, and try config/doc/kitti2bag.py like `python kitti2bag.py /path/to/kitti`.

closed 2020-11-20 by stale[bot]

415522099, 2020-12-22:
i have the same problem . $_$