TixiaoShan/LIO-SAM #148: lio_sam_imuPreintegration-1

he-guo, 2020-11-03:
hi, I have this problem: 
[lio_sam_imuPreintegration-1] process has died [pid 30652, exit code -6, cmd /home/fire/catkin_lio/devel/lib/lio_sam/lio_sam_imuPreintegration __name:=lio_sam_imuPreintegration __log:=/home/fire/.ros/log/a40f131e-1d9d-11eb-b2d7-cc3d826486ce/lio_sam_imuPreintegration-1.log].
log file: /home/fire/.ros/log/a40f131e-1d9d-11eb-b2d7-cc3d826486ce/lio_sam_imuPreintegration-1*.log
[image]
is my imu data wrong? can you give me some suggestions?

gaowc1990, 2020-11-04:
I got the same problem with you~

he-guo, 2020-11-04:
Maybe the pose has changed a lot, generally the pose is not set correctly. I use the original IMU data and will not report this error, but the data will be jittered too much, which will cause confusion in the mapping. This problem occurs when I use imu data synthesized by fusion data. The problem has not been solved yet, I hope you can tell me what progress you have.

closed 2020-11-21 by stale[bot]

oussama-chhoud, 2021-11-21:
i have the same problem.
there is a solution available? 
Thanks in advance

lonlonago, 2021-12-31:
I have solved this problem . lonlonago@foxmail.com @oussama-chhoud  @he-guo  @gaowc1990

narutojxl, 2022-01-04:
Maybe you can try [allan_variance_ros](https://github.com/ori-drs/allan_variance_ros) to give lio-sam proper noise param firstly :)

oussama-chhoud, 2022-01-10:
@lonlonago can you please tell us the solution ?
thanks in advance

DuyNhatNguyenIT, 2023-08-16:
```
cd ~/Downloads/gtsam-4.0.2/
cd build
cmake -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF ..
sudo make install -j8
```

mikolodz, 2024-04-17:
Using gtsam-4.0.2 is not helping in my case. I'm still facing the issue. It can be due to the fact that I'm using Isaac Sim and I'm not sure if it publishes the sensor readings properly. 10hz Ouster 3d lidar publishes more than 50 messages/sec, often multiple of them in a single nanosecond period. I think there might be some internal Isaac Sim problem causing this. Although, IDK how the LIO-SAM should react in such a case.