TixiaoShan/LIO-SAM #151: How does this compare with LeGO-LOAM?

soulslicer, 2020-11-19:
As in title

xieqi1, 2020-11-23:
loose coupling -> tightly coupling

soulslicer, 2020-11-23:
I dont understand what you mean by that? Is the ground plane constraint non existent in this one then?

xieqi1, 2020-11-24:
ground plane constraint in LeGO-LOAM only used in scan-to-scan-match odometry rather than scan-to-map mapping, it make odometry faster than LOAM.
while in LIO-SAM,tightly coupling Lidar-IMU fusion provide better odometry, so it does not need scan-to-scan match to provide odometry.

soulslicer, 2020-11-24:
Can I ask what scan to scan match means?

xieqi1, 2020-11-24:
There is a thread named odometry which provide initial guess for mapping thread, in LOAM or LeGO-LOAM, scan-to-scan match is used as odometry, which means use current scan and last scan to do scan matching, the result offers initial guess for mapping.

soulslicer, 2020-11-24:
okay, i was just concerned that it referred to each lidar column for each scanning angle. looks like my questions are answered thank you

closed 2020-11-24 by soulslicer