TixiaoShan/LIO-SAM #153: Custon kitti2bag.py not working

HMellor, 2020-11-20:
I am unable to create more bag files using the custom `kitti2bag.py` script. I am running noetic on Ubuntu 20.04 LTS with Python 3.8.5

`progressbar` was giving me issues, but I fixed them by switching to `tqdm`, this is superficial though. When I try and create more bag files using the process in the README file I get the following error:
```
Traceback (most recent call last):
  File "kitti2bag.py", line 422, in <module>
    save_velo_data(bag, kitti, velo_frame_id, velo_topic)
  File "kitti2bag.py", line 262, in save_velo_data
    pcl_msg = pcl2.create_cloud(header, fields, scan)
  File "/opt/ros/noetic/lib/python3/dist-packages/sensor_msgs/point_cloud2.py", line 164, in create_cloud
    pack_into(buff, offset, *p)
struct.error: required argument is not an integer
```
I am raising an issue here rather on the kitti2bag repo because I suspect it has something to do with the modifications to the LiDAR data

closed 2020-11-23 by HMellor

HappySamuel, 2021-02-23:
Hi @HMellor 

I saw that you're running LIO-SAM on the Ubuntu 20.04 with ROS Noetic. May i ask, how you compile the gtsam-4.0.2 on Ubuntu 20.04?

Best,
Samuel

HMellor, 2021-02-23:
Hi @HappySamuel, 

I've been using gtsam-4.0.3 built using [the build instructions on the gtsam website](https://gtsam.org/build/). The only change I had to make was to use `sudo make install` instead of `make install` because I had a permission denied error otherwise.

If you don't need to build from source, the [gtsam get started page](https://gtsam.org/get_started/) says that there is a published binary for Ubuntu 20.04, so you could use that instead.

AnarchistKnight, 2024-11-04:
good day, have the bug already been fixed? may I ask how?