TixiaoShan/LIO-SAM #154: How can I get trajectory??

minwo0jung, 2020-11-21:
It is known that LIO-SAM basically gives a trajectory point.
There is a trajectory comparison in the LIO-SAM paper. 
I want to see the trajectory as a line as shown in the picture. 
How can I achieve such a result? 

Thank you!

px4Li, 2020-12-02:
@minwo0jung convert the trajectory topic of rosbag to csv file.

closed 2020-12-18 by stale[bot]

LongruiDong, 2022-03-24:
> Hello. I am a student studying LIO-SAM.
> 
> It is known that LIO-SAM basically gives a trajectory point. However, there is a trajectory comparison in the LIO-SAM paper. I want to see the trajectory as a line as shown in the picture. How can I achieve such a result? Is there any other way besides saving the update pose values ​​directly in txt?
> 
> Thank you!

Hi~

have you figured out how to save final optimized pose?

LongruiDong, 2022-03-27:
> @minwo0jung convert the trajectory topic of rosbag to csv file.

Thans for your reply, actually I am new to ROS... :-)
could you explain more specifically that how to convert that rosbag to csv file?

minwoo0611, 2022-03-27:
> > @minwo0jung convert the trajectory topic of rosbag to csv file.
> 
> Thans for your reply, actually I am new to ROS... :-) could you explain more specifically that how to convert that rosbag to csv file?

Hello. I can give you two solution to get optimized trajectory.

Firstly, you can directly acheive trajectory. This method is easier and effective.
The optimized trajectory is published as "/lio_sam/mapping/odometry".
If you see the function "publishOdometry" in mapOptimization.cpp, there is laserOdometryROS and the value of that variable is published. You can save the value of the variable using std::ofstream. If you directly save the file such as " file << time << x << y << z ", there will be a problem with significant figure. So, I recommend to exploit file::precision function together.

Secondly, I recommend this method if you are unfamiliar with ROS. It may be a good example to understand subscriber and publisher.
Make your own cpp file to subscribe optimized topic and make callback function to save trajectory.

I hope these answers solve your problem clearly.

Thank you.

LongruiDong, 2022-03-28:
> Hello. I can give you two solution to get optimized trajectory.
>
> Firstly, you can directly acheive trajectory. This method is easier and effective. The optimized trajectory is published as [...]

what a great answer!
Thanks again~

farhad-dalirani, 2023-04-09:
@minwoo0611 I want to know the direction and position of the robot. I get points with /lio_sam/mapping/cloud_registered_raw; However, I do not know how to get the position and direction of the robot each time the pointcloud is published.