TixiaoShan/LIO-SAM #156: GTSAM from ppa

Timple, 2020-11-23:
The GTSAM [Quick-start guide](https://gtsam.org/get_started/) also states there is a ppa available.

It seems like it's using version [4.0.3](https://launchpad.net/~borglab/+archive/ubuntu/gtsam-release-4.0)

Would this work? I am not sure what the `-DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF` does and if that is also present in the ppa.

Timple, 2020-12-08:
I figured out that the `BUILD_WITH_MARCH_NATIVE` would need to be off for a ppa. So that is alright.

Now I have to find the time to test this :)

closed 2020-12-24 by stale[bot]

HappySamuel, 2021-02-23:
Hi @Timple 

I am currently also trying to use the gtsam (4.0.3) from ppa, how do you turn OFF the BUILD_WITH_MARCH_NATIVE? 

Best,
Samuel