TixiaoShan/LIO-SAM #162: KITTI Dataset

LeisureLei, 2020-12-02:
Hi  @TixiaoShan , I  noticed that you update KITTI data several days ago. And I tried new KITTI data. Lidar and IMU data are ok. But GPS data seems to have some problems.You can see below picture. The red path is the result of lio sam and another one is gps path tranformed to local cartesian coordinate.
Another question is lio sam will drift a lot without loop closure. Does IMU noises cause that ?
[image]

px4Li, 2020-12-02:
@LeisureLei Hi, I suspense you are set gpsTopic as "odometry/gpsz". Did you tried "odometry/gps"?

LeisureLei, 2020-12-02:
> @LeisureLei Hi, I suspense you are set gpsTopic as "odometry/gpsz". Did you tried "odometry/gps"?

Yse, I set gpsTopic as "odometry/gps",  but I get nothing from this topic. Also, I have already install robot localization package. Thus I use geographic package to tranform the message from "/gps/fix" to UTM coordinate(The result is shown above). Have you already tried new KITTI data with gps ? Is there anything OK ?

px4Li, 2020-12-03:
@LeisureLei I am not sure you are taking about kitti_2011_09_30_drive_0028_synced.bag. But I did't get any issue run this dataset.

LeisureLei, 2020-12-03:
> @LeisureLei I am not sure you are taking about kitti_2011_09_30_drive_0028_synced.bag. But I did't get any issue run this dataset.

Yse, when I was using kitti_2011_09_30_drive_0028_synced.bag to try lio sam, I found that lio sam has an obvious difference with grountTruth like below picture. Thus I tried to add GPS to constraint trajetory， but I got a bad result, which made me to check the GPS message. 
[image]
[image]

px4Li, 2020-12-03:
@LeisureLei Sorry I did not get your point which the lio-sam is obvious difference with ground truth. It is really close to ground truth as I saw it :). By the way could you tell me where you got the ground truth data? Is that from the same rosbag?

LeisureLei, 2020-12-03:
> @LeisureLei Sorry I did not get your point which the lio-sam is obvious difference with ground truth. It is really close to ground truth as I saw it :). By the way could you tell me where you got the ground truth data? Is that from the same rosbag?

The diffence is mainly on z-axis. 
The groundtruth data is from KITTI website——08.txt file.(The gps data from kitti_2011_09_30_drive_0028_synced.bag seems not to be used as groundtruth )

HMellor, 2020-12-11:
@LeisureLei Looking at the navsat launch file it appears that the gps odometry topic is called `odometry/navsat` ([source](https://github.com/TixiaoShan/LIO-SAM/blob/2af92ab288656fb157326c1e0a32b621ae183693/launch/include/module_navsat.launch#L17)), have you tried using that instead of `odometry/gps`?

closed 2020-12-26 by stale[bot]

yxallnyy, 2021-03-30:
@LeisureLei could you give me the link of kitti_2011_09_30_drive_0028_synced.bag ? thank you.

yxallnyy, 2021-04-06:
@px4Li  hi, could you show me that how you compare the lio-sam result with groundtruth ? evo_ape only supports three formats , kitti,tum&euroc. kitti format requires same poses size , tum requies simliar timestamps...i am confused about it. thanks.

LeisureLei, 2021-04-07:
> @px4Li hi, could you show me that how you compare the lio-sam result with groundtruth ? evo_ape only supports three formats , kitti,tum&euroc. kitti format requires same poses size , tum requies simliar timestamps...i am confused about it. thanks.

I use tum format.The groundtruth may have no timestamp.Thus I add timestamp manually.The real timestamp can get form bag.

yxallnyy, 2021-04-07:
> I use tum format.The groundtruth may have no timestamp.Thus I add timestamp manually.The real timestamp can get form bag.

would you mean that you transform the groundtruth(kitti format) to tum format, then compare?

LeisureLei, 2021-04-07:
> would you mean that you transform the groundtruth(kitti format) to tum format, then compare?

Yes, you can also compare trajectory using kitti format.Just saving lio-sam results as kitti format and changing absolute time into relative time

yxallnyy, 2021-04-07:
> Yes, you can also compare trajectory using kitti format.Just saving lio-sam results as kitti format and changing absolute time into relative time

i am still confused about it. would you add my qq(744895490) . thank you .

polbolso, 2021-04-29:
@LeisureLei Where precisely did you find the ground truth for sequence 2011_09_30_drive_0028_sync in KITTI's website? You mentioned something like ——08.txt file but I can't find it anywhere.

polbolso, 2021-05-18:
> Yse, when I was using kitti_2011_09_30_drive_0028_synced.bag to try lio sam, I found that lio sam has an obvious difference with grountTruth like below picture. Thus I tried to add GPS to constraint [...]

@LeisureLei I also used using kitti_2011_09_30_drive_0028_synced.bag for evaluation but it seems that my results are very misaligned in comparison with the ground truth. Have you experienced something similar in the beginning or do you have any idea on how to solve this issue?
[image]

polbolso, 2021-06-07:
Hi @LeisureLei I managed to align the results with the ground truth. But, there are still large differences (image below). I see that your results using the same dataset are quite good. Did you change anything else, like for example the IMU intrinsics?

[image]

LeisureLei, 2021-06-07:
> Hi @LeisureLei I managed to align the results with the ground truth. But, there are still large differences (image below). I see that your results using the same dataset are quite good. Did you change anything else, like for example the IMU intrinsics?
> 
> [image]

I haven't modefied IMU intrinsics. I use evo tools to align traj but with tum format, like "evo_traj tum groungTruth.txt test_traj.txt -va -p"

polbolso, 2021-06-07:
> I haven't modefied IMU intrinsics. I use evo tools to align traj but with tum format, like "evo_traj tum groungTruth.txt test_traj.txt -va -p"

Yes, I also did exactly the same as you say, but the results I got are the ones shown in the image. As you can see, they are quite different from the ones you obtained before, while they should be the same as we are using the same input (I only changed the extrinsic IMU-LiDAR as recommended). I can't think of something altering the results.

polbolso, 2021-06-18:
> I haven't modefied IMU intrinsics. I use evo tools to align traj but with tum format, like "evo_traj tum groungTruth.txt test_traj.txt -va -p"

Hi @LeisureLei, could you please share with me the final TXT file you used when comparing the LIO-SAM results in evo with KITTI's ground truth? Or send it through email to seccopablo@gmail.com.

I just need this to show as sample results obtained with this algorithm for my thesis. I tried a lot but couldn't solve/fix it myself, so I would really appreciate your help.

LeisureLei, 2021-06-23:
> > > Hi @LeisureLei I managed to align the results with the ground truth. But, there are still large differences (image below). I see that your results using the same dataset are quite good. Did you change anything else, like for example the IMU intrinsics?
> > > [image]
> > 
> > 
> > I haven't modefied IMU intrinsics. I use evo tools to align traj but with tum format, like "evo_traj tum groungTruth.txt test_traj.txt -va -p"
> 
> Hi @LeisureLei, could you please share with me the final TXT file you used when comparing the LIO-SAM results in evo with KITTI's ground truth? Or send it through email to [seccopablo@gmail.com](mailto:seccopablo@gmail.com).
> 
> I just need this to show as sample results obtained with this algorithm for my thesis. I tried a lot but couldn't solve/fix it myself, so I would really appreciate your help.

[kitti-loop.txt](https://github.com/TixiaoShan/LIO-SAM/files/6700116/kitti-loop.txt)
Sorry for reply late. I just noticed your message. Here is the txt file. But I am not sure whether it will satisfy you.

tigerschweiger, 2021-11-04:
> I use tum format.The groundtruth may have no timestamp.Thus I add timestamp manually.The real timestamp can get form bag.

hi！can you tell me how to add the timestamp? if possible could u send to me the gt file with timestamp?THX!tigerschweiger49@gmail.com

Diggle221, 2021-12-16:
> [kitti-loop.txt](https://github.com/TixiaoShan/LIO-SAM/files/6700116/kitti-loop.txt) Sorry for reply late. I just noticed your message. Here is the txt file. But I am not sure whether it will [...]

Hello, is the txt you provided groundtruth?

dxdydz3dt, 2023-06-24:
> Hi @TixiaoShan , I noticed that you update KITTI data several days ago. And I tried new KITTI data. Lidar and IMU data are ok. But GPS data seems to have some problems.You can see below picture. The [...]

can you share me ground truth files. 
dxdydz3dt@gmail.com

harisbinyousaf24, 2023-07-12:
Hey guys! @LeisureLei , @px4Li @yxallnyy @tigerschweiger Apologies...i am a bit late to the party..and i need some help regarding misalignment of imu preintegration based trajectory and GPS trajectory. This is the satellite view of both trajectories:
[image]
(if you are somehow not able to see the uploaded img then) Actually imu based trajectory is slightly off the road like tilted anticlockwise slightly wrt gps trajectory (ground truth). So can you tell me what changes i need to make in params.yaml file? One idea i got is to change the imu intrinsics (modify bias noises etc) but i want to know would that help?
Also one more thing, there are two trajectory topics published as a result of liosam /lio_sam/imu/path (Pubisher: imuPreintegraton module) and /lio_sam/mapping/path (Publisher: mapOptimization module), I want to know what is the actual; output trajectory of liosam? Apparently both trajectories overlap but are off-road wrt gps trajectory but still I need clarification.

harisbinyousaf24, 2023-09-21:
Hey @ataparlar I found its workaround not by tweaking parameters of LIO-SAM params.yaml but relocalized the output trajectory by integrating robot_localization package of ROS on output liosam trajectory and the gps sensor data. That was months ago but the output trajectory was much aligned with ground truth!. Hope that helps!

ashwinjoseph95, 2024-11-05:
Just wanted to enquire, what can be done about this warning message (except ignoring) to enable the deskew function:
"Point cloud timestamp not available, deskew function disabled, system will drift significantly!" i get while running LIO-SAM?

ashwinjoseph95, 2024-11-05:
Also while running with another rosbag I got from kitti raw say sequence 0027 and launching the lio sam launch file, I get this error message:
"[ERROR] [1730802813.546919026]: Point cloud is not in dense format, please remove NaN points first!"
how to fix this?

ataparlar, 2024-11-05:
Hi @ashwinjoseph95 

> Just wanted to enquire, what can be done about this warning message (except ignoring) to enable the deskew function: "Point cloud timestamp not available, deskew function disabled, system will drift significantly!" i get while running LIO-SAM?

You need to change the quote in [this line](https://github.com/TixiaoShan/LIO-SAM/blob/master/src/imageProjection.cpp#L273) with the timestamp field of your cloud.

> Also while running with another rosbag I got from kitti raw say sequence 0027 and launching the lio sam launch file, I get this error message: "[ERROR] [1730802813.546919026]: Point cloud is not in dense format, please remove NaN points first!" how to fix this?

And for this, you should check [this comment](https://github.com/TixiaoShan/LIO-SAM/issues/175#issuecomment-749456197).

ataparlar, 2024-11-05:
Hi @harisbinyousaf24 

> > Hey guys! @LeisureLei , @px4Li @yxallnyy @tigerschweiger Apologies...i am a bit late to the party..and i need some help regarding misalignment of imu preintegration based trajectory and GPS trajectory. This is the satellite view of both trajectories: [image] (if you are somehow not able to see the uploaded img then) Actually imu based trajectory is slightly off the road like tilted anticlockwise slightly wrt gps trajectory (ground truth). So can you tell me what changes i need to make in params.yaml file? One idea i got is to change the imu intrinsics (modify bias noises etc) but i want to know would that help? Also one more thing, there are two trajectory topics published as a result of liosam /lio_sam/imu/path (Pubisher: imuPreintegraton module) and /lio_sam/mapping/path (Publisher: mapOptimization module), I want to know what is the actual; output trajectory of liosam? Apparently both trajectories overlap but are off-road wrt gps trajectory but still I need clarification.
> 
> Hi, have you solved this problem? I am facing with a similar thing right now.

I found out that this problem is related to the coordinate conversion methods. If you are giving your GNSS sensor's output `Odometry` msg as the LIO-SAM's input `Odometry` ([gpsTopic](https://github.com/TixiaoShan/LIO-SAM/blob/master/config/params.yaml#L7) in the param file), there most likely be a rotation difference between the paths. Because sensors usually use `GeographicLib::LocalCartesian::Forward` for converting (lat, lon, alt) to (x, y, z).  You should use `GeographicLib::UTMUPS::Forward' instead as a position source.