TixiaoShan/LIO-SAM #165: Thrown when a linear system is ill-posed

he-guo, 2020-12-07:
Hello everyone,
    I have this problem when I rosbag my bag.
    
    terminate called after throwing an instance of 'gtsam::IndeterminantLinearSystemException'
  what():  
Indeterminant linear system detected while working near variable
7061644215716937728 (Symbol: b0).

Thrown when a linear system is ill-posed.  The most common cause for this
error is having underconstrained variables.  Mathematically, the system is
underdetermined.  See the GTSAM Doxygen documentation at
http://borg.cc.gatech.edu/ on gtsam::IndeterminantLinearSystemException for
more information.
[lio_sam_imuPreintegration-1] process has died [pid 31169, exit code -6, cmd /home/fire/catkin_lio/devel/lib/lio_sam/lio_sam_imuPreintegration __name:=lio_sam_imuPreintegration __log:=/home/fire/.ros/log/543bdf56-386a-11eb-a82b-cc3d826486ce/lio_sam_imuPreintegration-1.log].
log file: /home/fire/.ros/log/543bdf56-386a-11eb-a82b-cc3d826486ce/lio_sam_imuPreintegration-1*.log

Thanks very much if anyone gives some advice.

ajanani85, 2020-12-17:
Hi There, did you find any solution for this issue?

he-guo, 2020-12-21:
> Hi There, did you find any solution for this issue?

sorry, no. But I guess this problem is caused by  IMU raw data error.

closed 2021-01-06 by stale[bot]

xiaoxinnn, 2021-01-31:
when i run my own dataset, it comes to the same problem, but i check my imu gravity output, it looks right,  any suggestion to this issue?

rohansingh42, 2021-02-08:
Having the same issue. Did anyone fix this?

ajay1606, 2021-06-10:
Hello Everyone,
First of all thanks for sharing this open-source. 
AM also trying to run Lio-Sam with its own dataset, but getting the same error. Does anyone fix this before?

Regards,
Ajay

tigerschweiger, 2021-12-18:
Hey i have this problem because i mixed up the data from lidar odometry and visual odometry, when i motified them, everything was ok!

SlammerD, 2021-12-23:
I had this problem too. In my case the reason was that some imu messages were sent with the same timestamp. This causes trouble in imuPreintegration.cpp lines 356, 424 and 469. As you can see there is no check if the value is equal to 0. So in my case dt gets 0 which then in turn leads to the fact that in gtsam the systems of equations can no longer be solved.

Alex-Beh, 2022-02-24:
@SlammerD Can you briefly elaborate more on this issue?

I did you what you said in this [comment](https://github.com/TixiaoShan/LIO-SAM/issues/303#issuecomment-1005960133) but still facing the same issue when I run the mapping on the example bagfile. Do I need to do the same change to lines 424 & line 469?

SlammerD, 2022-02-25:
> I did you what you said in this [comment](https://github.com/TixiaoShan/LIO-SAM/issues/303#issuecomment-1005960133) but still facing the same issue when I run the mapping on the example bagfile. Do I need to do the same change to lines 424 & line 469?
When you have trouble with the example bagfile than you don't have my problem. Therefore you don't need to change anything in the code of lio sam. The example bagfile should work right of the box. Only slight changes in the params file are needed for some examples but they are mentioned in the explanation.