TixiaoShan/LIO-SAM #167: Difficulty setting up LIO-SAM for custom rosbag - "Not enough features! Only 0 edge and 0 planar features available"

WysokiStudent, 2020-12-08:
After launching LIO-SAM with `roslaunch lio_sam run.launch` and playing a rosbag with `rosbag play long_walk.bag`. The roll and pitch angles are off from what I expected. I'm not sure what is going on and would appreciate any advice.

The test data was gathered by making a short loop around a building on my university campus. An IMU and lidar were mounted to a box that was handheld above my head. The IMU was UM-7 which is a 9-axis IMU and the lidar was a VLP-32C scanner. The IMU is mounted under the lidar upside down, the axis are shown in the following picture. Red - X, Green - Y, Blue - Z.
[image]

An approximate of the path taken can be seen in the following picture. I moved clockwise along the blue line. The red dot represents the starting and final position.

[image]

The roll and pitch of the device used should be close to 0 during the whole walk but they look like this:
```
IMU acc: 
x: 0.642488, y: 0.532171, z: 10.0666
IMU gyro: 
x: 0.0160344, y: 0.00455378, z: 0.0205549
IMU roll pitch yaw: 
roll: 0.440268, pitch: -1.42348, yaw: -1.92964
````
To transform IMU acceleration and angular velocity data from the IMU frame to the device frame the following calculations are used:
acc_dev_3x1 = RdevIMU * MKA * [acc_imu_3x1, 1]
gyro_dev_3x1 = RdevIMU * MKG * [gyro_imu_3x1, 1]
where RDevIMU is a rotation matrix, MKA is a calibration matrix and so is MKG.
I upload a script where this transformation is done. The IMU used has inverted Y and Z axis, both for the accelerometer and the gyroscope.  In the script the acceleration and angular velocity are transformed into the device frame accounting for this but not the quaternion for which I use the following rotation parameters in params.yaml:
```
extrinsicRot: 
[1, 0, 0,
 0, 1, 0,
 0, 0, 1]
extrinsicRPY:
[0, 1, 0,
 1, 0, 0,
 0, 0, -1]
```
From my understanding this should give a correct rotation. This setup will also sometimes crash imuPreintegration which will die after an exception is thrown.
```
terminate called after throwing an instance of 'gtsam::IndeterminantLinearSystemException'
  what():  
Indeterminant linear system detected while working near variable
7061644215716937728 (Symbol: b0).

Thrown when a linear system is ill-posed.  The most common cause for this
error is having underconstrained variables.  Mathematically, the system is
underdetermined.  See the GTSAM Doxygen documentation at
http://borg.cc.gatech.edu/ on gtsam::IndeterminantLinearSystemException for
more information.
```
My questions are the following:
1. Why is this exception thrown?
2. Is my extrinsicRPY incorrect?

params.yaml:
```
lio_sam:

  # Original Topics
  # pointCloudTopic: "points_raw"               # Point cloud data
  # imuTopic: "imu_raw"                         # IMU data
  # odomTopic: "odometry/imu"                   # IMU pre-preintegration odometry, same frequency as IMU
  # gpsTopic: "odometry/gpsz"                   # GPS odometry topic from navsat, see module_navsat.launch file

  # My Topics
  pointCloudTopic: "velodyne_points"          # Point cloud data
  imuTopic: "calibrated_imu"                  # IMU data
  odomTopic: "odometry/imu"                   # IMU pre-preintegration odometry, same frequency as IMU
  gpsTopic: "odometry/gps"                   # GPS odometry topic from navsat, see module_navsat.launch file

  # Frames
  lidarFrame: "base_link"
  baselinkFrame: "base_link"
  odometryFrame: "odom"
[... 209 lines cut ...]

module_navsat:
```
<launch>

    <arg name="project" default="lio_sam"/>

    <env name="ROSCONSOLE_CONFIG_FILE" value="$(find lio_sam)/launch/include/rosconsole/rosconsole_error.conf"/>
    
    <!-- EKF GPS-->
    <node pkg="robot_localization" type="ekf_localization_node" name="ekf_gps" respawn="true">
        <remap from="odometry/filtered" to="odometry/navsat" />
    </node>

    <!-- Navsat -->
    <node pkg="robot_localization" type="navsat_transform_node" name="navsat" respawn="true">
        <!-- <rosparam param="datum">[42.35893211, -71.09345588, 0.0, world, base_link]</rosparam> -->
        <remap from="imu/data" to="imu_correct" />
        <remap from="fix" to="gps/fix" />
        <remap from="odometry/filtered" to="odometry/navsat" />
    </node>

</launch>
```
Link to rosbag: https://drive.google.com/file/d/1Ms5_R4K6w-95e5mpBkkbODRiTuqrOWbj/view?usp=sharing

closed 2020-12-24 by stale[bot]

aniltotakura, 2021-08-10:
Hai, I have the same problem. How did you solve it?. Did you try to changing the function `imuConverter(const sensor_msgs::& imu_in )`. Here the line where the `q_final` is modified by the multiplication of quaternion from the extrinsicRPY parameter. 
Originally the code is `Eigen::Quaterniond q_final = q_from * extQRPY;` did you change it to 
`Eigen::Quaterniond q_final = extQRPY * q_from * extQRPY.inverse();`. This does nothing for me.

WysokiStudent, 2021-08-10:
Hi @aniltotakura, I'm sorry to say but I didn't find a solution. We managed to get LIO-SAM working eventually by:
* Using a higher quality IMU
* Aligning all axis of IMU and Lidar
* Ignoring GPS
* Using a more powerful PC for running LIO-SAM (It's not real-time on a laptop)

Please also note that LIO-SAM is very susceptible to getting lost when the device used to gather data is being shaken. It also does not deal well with data discontinuities.

Additionally note that the ROS driver for Velodyne Lidars' returns data in a different coordinate frame than the one specified in Velodyne's manuals.

I wish you luck and patience.

SlammerD, 2021-12-23:
First of all I wonder why your coordinate system is left handed, but anyway. After I failed with giving the imu data already correctly transformed into lio sam I tried it with only making the right specifications in the params.yaml. Maybe it would work for you too. Let out your own transformation and just match the frames in params.yaml. I would try it with this:

extrinsicRot: 
[0, 1, 0,
 -1, 0, 0,
 0, 0, 1]
extrinsicRPY:
[0, 1, 0,
 -1, 0, 0,
 0, 0, 1]

For your problem with the crash of imuPreintegration you can have a look here: https://github.com/TixiaoShan/LIO-SAM/issues/165#issuecomment-1000247782