TixiaoShan/LIO-SAM #169: how to get extrinsic and time offset between imu and lidar ?

chennuo0125-HIT, 2020-12-14:


HMellor, 2020-12-14:
Extrinsic offset is measured (by hand or using CAD if you designed a mounting bracket) between the sensor output origin for your IMU and LiDAR. Sensor origins can be found in the manuals for your sensors.

What do you mean by time offset?

chennuo0125-HIT, 2020-12-15:
measure extrinsic by hand isnt precise, this error will influence mapping precise ?

chennuo0125-HIT, 2020-12-15:
time offset means time difference introduced by clock out of sync

HMellor, 2020-12-15:
> measure extrinsic by hand isnt precise, this error will influence mapping precise ?
You're right. But unless you have CAD models and CNC/3D printed mounting hardware that is the best you can do with the current implementation. You could try to optimise the extrinsic values based on the quality of the mapping, but that is not trivial or part of this codebase.

> time offset means time difference introduced by clock out of sync
As far as I'm aware (@TixiaoShan correct me if I'm wrong), there are no parameters for synchronising the clocks of the sensors. I believe that if the sensors are connected to the same computer as the ROS master, they will all be operating from the system clock.

chennuo0125-HIT, 2020-12-16:
i have wrote a project to calibrate the extrinsic rotation from lidar to imu [https://github.com/chennuo0125-HIT/lidar_imu_calib](https://github.com/chennuo0125-HIT/lidar_imu_calib)

chennuo0125-HIT, 2020-12-16:
measure extrinsic by hand is Inappropriate, especially for product development, everyone maybe get different result :(

closed 2021-01-02 by stale[bot]