TixiaoShan/LIO-SAM #171: Oscillating altitude when LiDAR+IMU stationary

HMellor, 2020-12-14:
# Problem
Pose solution is slowly oscillating up and down centred on the initial origin of the global map.

The period is ~10s so this is a very slow drift. It appears to be an issue with having the value of gravity set as a static parameter. If I set gravity to be a smaller value, say `9.77`, then the pose drifts upwards initially and then begins to oscillate as the LiDAR factor appears to constrain the infinite drift. Though, this could be an incorrect hypothesis. @TixiaoShan do you know of a good way to infer the value of g that would avoid this issue?

# Steps to reproduce
* Disable loop closure
* Set gravity to `9.77`
* Start LIO-SAM and watch what happens

TixiaoShan, 2020-12-14:
Please refer to this issue to get the correct parameters for IMU.

https://github.com/TixiaoShan/LIO-SAM/issues/91

HMellor, 2020-12-14:
Thanks for the quick response, I have actually already used that library to get the noise and bias noise values (Image from #91) [image]

But I am still experiencing this oscillation.

TixiaoShan, 2020-12-14:
The drift might be because there are not enough keyframes used to generate a map for scan-matching. Try to move the sensor around and maybe make some changes to [this function](https://github.com/TixiaoShan/LIO-SAM/blob/a53730e96913024445d53c4f5d2e4ea875a14b5e/src/mapOptmization.cpp#L811)

celikemir, 2020-12-14:
@TixiaoShan Does it make sense to use imu_utils lib to set correct IMU settings params because it needs 2 hours data and imu should be Stationary. Did you get those params using imu_utils ?

TixiaoShan, 2020-12-14:
@celikemir 
Yes, I did.

closed 2020-12-31 by stale[bot]

HMellor, 2021-03-16:
For the benefit of anyone who stumbles across this in the future (@shibowing), I now have some idea as to why the oscillation was occurring:
* Take the situation where there are no features to constrain the LiDAR in a given direction (i.e. you are in a room with perfectly smooth walls and the LiDAR cannot see the ceiling or the floor, meaning that there is no vertical constraint)
* In this situation the only thing that can constrain LIO-SAM's vertical position is the IMU, and if you have entered an incorrect value of gravity in `params.yaml` then you are going to start with a large IMU bias, causing LIO-SAM to move up or down
* I suspect that the oscillation is the LIO-SAM frequency response of LIO-SAM optimising this bias (i.e. it sinusoidally approaches the correct value)