TixiaoShan/LIO-SAM #172: Drift

celikemir, 2020-12-14:
Hi 

Could you look up the attached gift which shows drift. What do you think is the reason? IMU and lidar stuck together very well and during that time they were motionless. When I move  lidar and imu like your walking data set ( I also watched on Youtube) it doesn't give as good as your result

[image]

HMellor, 2020-12-14:
Hi, this is a very similar issue to #171, does the pose drift back to the world origin if you leave it long enough?

celikemir, 2020-12-14:
sometimes it happens but %90 there is no retrieve to do beginning original pose  its like a cumulative error.
If I move lidar+ımu the result like this. I will update imuAccNoise GyrNoise matrices by using your issue Github code reference. But Should I rewrite process_noise_covariance matrice ? If yes , How :)
[image]

rohansingh42, 2020-12-28:
Hi,
I am running the code on real-time data with an ouster lidar mounted on a UAV and having similar problems. Did you figure out any fix for this?
Thanks

celikemir, 2020-12-28:
Ouster has different settings be sure you set them correctly. I solved the drift doing lidar imu calibration.

rohansingh42, 2020-12-28:
Yeah, I'm pretty sure I set everything correctly. I used the imu calibration package and made sure the frames and os gen is correct(gen2). Although I didn't use the lidar-imu calib package. Is that what you meant?

HMellor, 2020-12-29:
In my case at least some of the issues were down to a timing issue. If your IMU is connected via USB, make sure that you're using low latency mode ([for Windows](https://base.xsens.com/hc/en-us/community/posts/360029341694-How-do-I-get-IMU-data-via-USB-in-real-time-?page=1#community_comment_360004975314) and [for Unix](https://stackoverflow.com/a/13136412/10880714)

closed 2021-01-16 by stale[bot]

andu215, 2021-06-01:
> Ouster has different settings be sure you set them correctly. I solved the drift doing lidar imu calibration.

Hi, 
I have the same problem as you @celikemir.
Could you describe in more detail how you went about solving the problem?

If I use this Lidar-IMU-Calibration: https://github.com/chennuo0125-HIT/lidar_imu_calib 
and adjust the extrinsicRot, no big changes occur compared to [1,0,0; 0,1,0; 0,0,1].

I use a VLP-16 and an XSense MTI-G-700.

Thank you in advance

yuanguobin, 2022-03-28:
> Hi, I have the same problem as you @celikemir. Could you describe in more detail how you went about solving the problem?
>
> If I use this Lidar-IMU-Calibration: [...]

I think that because there is no constraint on Z indoor，You can try set the initial Z to 0。