TixiaoShan/LIO-SAM #174: the velodyne ros package does not have time.

he-guo, 2020-12-21:
Hi, I used the VLP-16 lidar and the point cloud message velodyne_points sent by the ROS drive software package directly, which only contained ring without time. Did I make a mistake?

narutojxl, 2020-12-21:
please update the vlp-16 ros driver to the latest, you will get time.

he-guo, 2020-12-21:
> please update the vlp-16 ros driver to the latest, you will get time.

Oh, yes, this is the VLP-16 ros driver I downloaded before. Thank you very much for your information

closed 2021-01-06 by stale[bot]

FengJianLL, 2021-03-28:
> > please update the vlp-16 ros driver to the latest, you will get time.
> 
> Oh, yes, this is the VLP-16 ros driver I downloaded before. Thank you very much for your information

Hi, I meet the same problem and I update the driver to the latest, but I still can not get time. Can you share your driver? Thank you!

FengJianLL, 2021-03-28:
> please update the vlp-16 ros driver to the latest, you will get time.

Hi, I meet the same problem and I update the driver to the latest, but I still can not get time. Can you share your driver? Thank you!

he-guo, 2021-04-30:
GitHub - ros-drivers/velodyne: ROS support for Velodyne 3D LIDARs

发送自 Windows 10 版邮件应用

发件人: FengJianLL
发送时间: 2021年3月28日 17:21
收件人: TixiaoShan/LIO-SAM
抄送: he-guo; Author
主题: Re: [TixiaoShan/LIO-SAM] the velodyne ros package does not havetime. (#174)

please update the vlp-16 ros driver to the latest, you will get time.
Hi, I meet the same problem and I update the driver to the latest, but I still can not get time. Can you share your driver? Thank you!