TixiaoShan/LIO-SAM #175: Point cloud is not in dense format, please remove NaN points first!

415522099, 2020-12-22:
I used RSLIDAR to make bag, and modified config/params.yaml pointcoudtopic. When I run this bag with lio-sam, the following problem occur.

[ERROR] [1608623674.483017189]: Point cloud is not in dense format, please remove NaN points first!

 How to solve this problem, please

Pallav1299, 2020-12-22:
Your pointcloud has NaN points. You need to clean your pointcloud. Add the following to imageProjection.cpp node:

```
        // remove Nan
        vector<int> indices;
        pcl::removeNaNFromPointCloud(*laserCloudIn, *laserCloudIn, indices);

        // check dense flag
        if (laserCloudIn->is_dense == false)
        {
            ROS_ERROR("Point cloud is not in dense format, please remove NaN points first!");
            ros::shutdown();
        }
```
This should work.

415522099, 2020-12-23:
> Your pointcloud has NaN points. You need to clean your pointcloud. Add the following to imageProjection.cpp node:
>
> ```
>         // remove Nan
>         vector<int> indices; [...]

> Your pointcloud has NaN points. You need to clean your pointcloud. Add the following to imageProjection.cpp node:
>
> ```
>         // remove Nan
>         vector<int> indices; [...]

Hi, I tried, I added this code to LIO-SAM/src/imageProjection.cpp, but it did not work.

Pallav1299, 2020-12-23:
Did you get any errors? Please try to provide some detailed information for reference.

415522099, 2020-12-24:
> Did you get any errors? Please try to provide some detailed information for reference.

 I have solved the previous problems by other methods, but I have encountered new problems 

`[ WARN] [1608712876.194493172]: Point cloud timestamp not available, deskew function disabled, system will drift significantly!
`
sad

Pallav1299, 2020-12-24:
It's a warning. Were you able to get LIO-SAM running? 
The algorithm expects a custom XYZIRT pointtype for LIDAR pointcloud, where point timestamp is used for deskewing the pointcloud.

zcg3648806, 2020-12-24:
> > Did you get any errors? Please try to provide some detailed information for reference.
> 
> I have solved the previous problem by other methods, but I have encountered new problems
> 
> `[ WARN] [1608712876.194493172]: Point cloud timestamp not available, deskew function disabled, system will drift significantly! `
> sad

Hello! Could you tell me what method you use to solve the NaN points problem?

heirenlop, 2020-12-24:
> Hello! Could you tell me what method you use to solve the NaN points problem?

[](url)

> Hello! Could you tell me what method you use to solve the NaN points problem?

I downloaded the latest Robesense driver. The new driver eliminates the Nan point cloud and has XYZIRT coordinates,which R is "ring" and T is "timestamp".
link is [https://github.com/RoboSense-LiDAR/rslidar_sdk](url)

zcg3648806, 2020-12-24:
> Your pointcloud has NaN points. You need to clean your pointcloud. Add the following to imageProjection.cpp node:
>
> ```
>         // remove Nan
>         vector<int> indices; [...]

Hi, After I add the above code, I get this error：
Undefined reference to void pcl::removeNaNFromPointCloud<PointXYZIRT>(pcl::PointCloud<PointXYZIRT> const&, pcl::PointCloud<PointXYZIRT>&, std::vector<int, std::allocator<int> >&)
Why？

zcg3648806, 2020-12-24:
> > > > Did you get any errors? Please try to provide some detailed information for reference.
> > > 
> > > 
> > > I have solved the previous problem by other methods, but I have encountered new problems
> > > `[ WARN] [1608712876.194493172]: Point cloud timestamp not available, deskew function disabled, system will drift significantly! `
> > > sad
> > 
> > 
> > Hello! Could you tell me what method you use to solve the NaN points problem?
> 
> > > > Did you get any errors? Please try to provide some detailed information for reference.
> > > 
> > > 
> > > I have solved the previous problem by other methods, but I have encountered new problems
> > > `[ WARN] [1608712876.194493172]: Point cloud timestamp not available, deskew function disabled, system will drift significantly! `
> > > sad
> > 
> > 
> > Hello! Could you tell me what method you use to solve the NaN points problem?
> 
> I downloaded the latest rslidar_sdk. The new driver eliminates the Nan point cloud and has XYZIRT coordinates,which R is "ring" and T is "timestamp".
> link is [https://github.com/RoboSense-LiDAR/rslidar_sdk](url)

I see, but I have been unable to collect any more data recently for some reason :(

zcg3648806, 2020-12-24:
> I downloaded the latest rslidar_sdk. The new driver eliminates the Nan point cloud and has XYZIRT coordinates,which R is "ring" and T is "timestamp".
> link is [...]

Excuse me, after you add the code：
        vector<int> indices;
        pcl::removeNaNFromPointCloud(*laserCloudIn, *laserCloudIn, indices);
is there no error in the compilation?

closed 2021-01-19 by stale[bot]

beyondpzk, 2021-09-12:
> > Your pointcloud has NaN points. You need to clean your pointcloud. Add the following to imageProjection.cpp node:
> > ```
> >         // remove Nan
> >         vector<int> indices;
> >         pcl::removeNaNFromPointCloud(*laserCloudIn, *laserCloudIn, indices);
> > 
> >         // check dense flag
> >         if (laserCloudIn->is_dense == false)
> >         {
> >             ROS_ERROR("Point cloud is not in dense format, please remove NaN points first!");
> >             ros::shutdown();
> >         }
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > This should work.
> 
> Hi, After I add the above code, I get this error：
> Undefined reference to void pcl::removeNaNFromPointCloud(pcl::PointCloud const&, pcl::PointCloud&, std::vector<int, std::allocator >&)
> Why？

it should be `std::vector`

z1047941150, 2021-12-28:
> 
It still wrong

z1047941150, 2021-12-28:
> it should be `std::vector`

After add std::vector, still not working

luameows, 2021-12-28:
> 

u need to add `#include "pcl/filters/impl/filter.hpp` first

luameows, 2021-12-28:
> After add std::vector, still not working

@z1047941150 u need to add `#include "pcl/filters/impl/filter.hpp"` first

z1047941150, 2021-12-29:
Thanks dude, I wil try it later.
 

Zhihao Zhang
***@***.***

     

         // remove Nan         vector<int&gt; indices;         pcl::removeNaNFromPointCloud(*laserCloudIn, *laserCloudIn, indices);         // check dense flag         if (laserCloudIn-&gt;is_dense == false)         {             ROS_ERROR("Point cloud is not in dense format, please remove NaN points first!");             ros::shutdown();         }  
  
 Undefined reference to void pcl::removeNaNFromPointCloud(pcl::PointCloud const&amp;, pcl::PointCloud&amp;, std::vector<int, std::allocator &gt;&amp;)

z1047941150, 2021-12-29:
> @z1047941150 u need to add `#include "pcl/filters/impl/filter.hpp"` first

Thanks, dude. It works， but I have a question. 
Lego-loam used this function too, why could work without adding  #include "pcl/filters/impl/filter.hpp"?

zcg3648806, 2021-12-31:
> Thanks, dude. It works， but I have a question. Lego-loam used this function too, why could work without adding #include "pcl/filters/impl/filter.hpp"?

Perhaps Lego-loam includes other libraries.

DaddyWesker, 2022-01-31:
Have the same issue. Adding 
```
#include "pcl/filters/impl/filter.hpp"
vector<int> indices;
pcl::removeNaNFromPointCloud(*laserCloudIn, *laserCloudIn, indices);
```
helped a bit, but after that i'm getting this 
`[ERROR] [1643635462.415513074]: Point cloud ring channel not available, please configure your point cloud data!`
Any clues?

arjunskumar, 2022-02-21:
@DaddyWesker are you able to solve the ring issue?

DaddyWesker, 2022-02-21:
@arjunskumar  unfortunately, no

z1047941150, 2022-02-22:
> 

Ring means the order of the beams. if your lidar is vlp-32, you will get 32 beams and each point will add the data which beam it belongs to. But some lidar didn't add this data to echo point, like RS-Lidar-32. So we should use other program to modify it, for example:https://github.com/EpsAvlc/rslidar_to_velodyne

z1047941150, 2022-02-22:
@DaddyWesker Same with the  please remove NaN points first problem, it because of  the different lidar data formats of the different lidar manufacturer.

z1047941150, 2022-02-22:
@DaddyWesker @arjunskumar 
Ring means the order of the beams. if your lidar is vlp-32, you will get 32 beams and each point will add the data which beam it belongs to. But some lidar didn't add this data to echo point, like RS-Lidar-32. So we should use other program to modify it, for example:https://github.com/EpsAvlc/rslidar_to_velodyne

arjunskumar, 2022-02-22:
@z1047941150 yes I manged to get the ring info , but not able to run on my dataset as IMU and Lidar were not calibrated.

z1047941150, 2022-02-22:
@arjunskumar well... Maybe because of the rate of IMU message... I didn't calibrate my IMU and Lidar too, but the program worked well.

arjunskumar, 2022-02-22:
@z1047941150 have you used lidar from robosense or the velodyne one?

z1047941150, 2022-02-22:
@arjunskumar yes,RS-Helios-32

arjunskumar, 2022-02-22:
if you dont mind, could you share your bag file @z1047941150 ?

arjunskumar, 2022-02-22:
@z1047941150 I have the same lidar with me.  It would be really helpful to see your configuration

z1047941150, 2022-02-22:
@arjunskumar I‘m sorry. Due to work regulations, I can't share it :(

arjunskumar, 2022-02-22:
@z1047941150 i can see [this code](https://github.com/EpsAvlc/rslidar_to_velodyne/blob/cfd1450cf82661882a363fec088de191302f13d9/src/convert.cpp#L77) uses 16 beams, if I change it to 32 and give those information to LIO-SAM, RS-Helios-32 should work right?

z1047941150, 2022-02-22:
@arjunskumar no, you should change the params.yaml.

DaddyWesker, 2022-02-22:
@z1047941150 I'm using simple Kitti dataset. As i know, it's velodyne.

arjunskumar, 2022-02-22:
@z1047941150 I tried converting the lidar but IMU outputs only at 50Hz. The author has recommended at least use 200 Hz IMU. Can you check the frequency of your IMU and update me ?

Attaching my output for your reference
[image]

z1047941150, 2022-02-22:
@arjunskumar  when I use IMU at 10HZ, it's output as same as yours and had bad performance when the robot tunring. Now I using 200hz, the output is very perfect.

z1047941150, 2022-02-22:
@DaddyWesker Maybe something wrong in the params.yaml?

harisbinyousaf24, 2023-06-07:
@z1047941150 Can you tell me what changings are needed to be made in the params.yaml file? should i need to change my topics names only? and i want to know where is this odometry/imu coming from? is this the topic published as a result of LIO-SAM? Are there any changings need to be made in .cpp files?  I am using same RS Helios 5515.

harisbinyousaf24, 2023-06-07:
@z1047941150 Thanks for the reply! Actually my data is coming from a rosbag file and there i have tf, imu. and rslidar topics. The data is recorded in real time and i want to know how to modify the package for my usage? Do i have to change the .cpp files? I couldn't understand the scripts for param.yaml. Would highly appreciate if you could help me

z1047941150, 2023-06-07:
> @z1047941150 Thanks for the reply! Actually my data is coming from a rosbag file and there i have tf, imu. and rslidar topics. The data is recorded in real time and i want to know how to modify the [...]

nothing need to be changed in package or .cpp files. The only thing you need to do is just convert it.

harisbinyousaf24, 2023-06-07:
Okay I will try and will let u know if any problem occurred. Thanks so much again for the help!!

harisbinyousaf24, 2023-06-07:
@z1047941150 So after converting i would be getting my data on this topic /velodyne_points So wouldn't i need to change points_raw topic or imu etc in the package Like my point cloud topic would become /velodyne_points and imu topic would become imu/data because thats the topic my rosbag is publishing on.

harisbinyousaf24, 2023-07-17:
> Thanks, dude. It works， but I have a question. Lego-loam used this function too, why could work without adding #include "pcl/filters/impl/filter.hpp"?

Hey @z1047941150 can you tell me how did you make that work? I added those lines in image Projection.cpp module but i am still gfetting this error:
[ERROR] [1689602770.451085416]: Point cloud is not in dense format, please remove NaN points first!

JzHuai0108, 2023-08-05:
Try

TZECHIN6, 2023-09-26:
The latest [rslidar_sdk](https://github.com/RoboSense-LiDAR/rslidar_sdk/blob/main/doc/howto/05_how_to_change_point_type.md) include `ring` and `timestamp` data field, you might have to rename `timestamp` to `time`, I am not sure is the `ring` sharing the same meaning.

For NaN issue, you might try setting `dense_points = true` as shown [here](https://github.com/RoboSense-LiDAR/rslidar_sdk/blob/main/doc/intro/02_parameter_intro.md#22-lidar)