TixiaoShan/LIO-SAM #178: lio_sam_imageProjection-3 died with "boost: mutex lock failed in pthread_mutex_lock: Invalid argument"

xiaodidia, 2020-12-28:
Hello Tixiao,
       lio_sam_imageProjection-3 died and restart, i paste the output of terminal below.
<<
       [lio_sam_imageProjection-3] restarting process
      process[lio_sam_imageProjection-3]: started with pid [31572]
      [ INFO] [1609152159.114445213]: ----> Image Projection Started.
      terminate called after throwing an instance of 
      'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
      what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
>>
     what might cause this kind of error in imageProjection process? please help.

TixiaoShan, 2020-12-29:
I didn't encounter that problem. You can try to put some ```cout``` function at different places of the code and see where the code breaks.

xiaodidia, 2021-01-11:
Thank you for your reply. That is the only time i encountered the problem. And everything seems to be alright with "cout" info. I am closing it.

closed 2021-01-11 by xiaodidia