TixiaoShan/LIO-SAM #179: mappingProcessInterval variable never used

rohansingh42, 2020-12-28:
Hi,
Thanks for sharing this code.
I have been going through the code and I came across the mappingProcessInterval variable which doesn't seem to be used after this commit [b4a1385](https://github.com/TixiaoShan/LIO-SAM/commit/b4a1385a77544e283aa82a34f576edccb12c1920). Is there any particular reason for this?

Thanks

TixiaoShan, 2020-12-29:
It's used in mapOptmization.cpp

rohansingh42, 2020-12-29:
I meant it's there but logically it shouldn't have any effect as the variable timeLastProcessing is being set to a constant value of -1 each time before the if statement(line 241) in function laserCloudInfoHandler. Please correct me if I missed anything. In your earlier commit as I mentioned this was set to -1 only once in class declaration.
`        static double timeLastProcessing = -1;
        if (timeLaserInfoCur - timeLastProcessing >= mappingProcessInterval)`

closed 2021-01-16 by stale[bot]