TixiaoShan/LIO-SAM #184: how to do localization with Lio slam generated map?

Thedush, 2021-01-06:
Hi currently i was working with the lio slam package but i want to know. is it possible to run the localization alone with a build map.
For example i have created a map using lio slam. now i want to localize my robot inside the map with build map features and graph.
is there anyway i can achieve this

xieqi1, 2021-01-06:
If you only save the global map, you can use general scan matching methods such as GICP or NDT.While if you save the map consist of corner point cloud and surf point cloud, you can also use the scan matching method just like what LOAM does.

Thedush, 2021-01-14:
How do i save the map generated by the lio-sam . In rviz they map was shown in '/lio_sam/mapping/cloud_registered' topic but it wasn't a complete map it was decayed points. how do i save the map to a pcd

closed 2021-01-14 by Thedush

xieqi1, 2021-01-15:
i think you can refer this issue #3

jjbecomespheh, 2021-01-29:
You can also use this pull request by zang09
https://github.com/TixiaoShan/LIO-SAM/pull/196