TixiaoShan/LIO-SAM #186: Can not run sample datasets

fergcd, 2021-01-08:
[image]

When I "roslaunch lio_sam run.launch", lio_sam_rviz is showing Global Status, base_link, and Velodyne "RED" on its Displays view.
And when I tried to run casual_walk.bag file, lio_sam_mapOptimization keeps dying and nothing is displayed on lio_sam_rviz.

Why is this happening and what should I do?  I installed gtsam following README.md, so I don't think it's gtsam version problem.
I'm using Ubuntu16.04 on VirtualBox.

cumulo-autumn, 2021-01-23:
I have the same problem. Any update?

px4Li, 2021-01-24:
did you install all of these:

sudo apt-get install -y ros-kinetic-navigation
sudo apt-get install -y ros-kinetic-robot-localization
sudo apt-get install -y ros-kinetic-robot-state-publisher

fergcd, 2021-01-25:
@px4Li Yes, I did.  I followed all the commands written in README.md.

cumulo-autumn, 2021-01-26:
I am using melodic. Should I still install ros-kinetic-navigation or should I install ros-melodic-navigation instead?

fergcd, 2021-01-27:
Problem solved by setting Virtual Machine's RAM from 4GB to 16GB.
I think it was caused by memory issue.

closed 2021-01-27 by fergcd