TixiaoShan/LIO-SAM #188: OS1-128 mapping error

JJuhyeon-Kim, 2021-01-13:
Dear TixiaoShan, thank you for your brilliant work.

I tried SLAM with Velodyne lidar using your source code and succeeded.
Now I'm trying to use Ouster lidar but there was a mapping error.
I used the same sensor as you. (OS1-128, LORD IMU sensor)

What I changed in 'params.yaml' file is topic name, lidar model, number of channel, and horizontal resolution. 
[image]

When I did roslaunch with ouster launch file, I defined timestamp mode as you mentioned. 
[image]

I don't know the cause, but I can't map it. I've been trying to solve this problem for weeks. If you know anything about this, give me some advice.
[image]

I also share my bag file here.
https://drive.google.com/drive/folders/1F_kMXPAdxfhmymauiMEeppUkxK9Gc26d?usp=sharing

px4Li, 2021-01-23:
LIO-SAM only works with a 9-axis IMU.

JJuhyeon-Kim, 2021-01-25:
> LIO-SAM only works with a 9-axis IMU.

Yes. I used 9-axis IMU, but I failed mapping with OS1-128

closed 2021-02-14 by stale[bot]

IvanVN, 2022-07-12:
Hi,

We are having what it looks like the same problem. We are doing the same as @JJuhyeon-Kim, with the same result.

I'm using the same Ouster OS1-128 with a Vectornav VN-100 IMU. We are using the [official Ouster driver](https://github.com/ouster-lidar/ouster_example).

We have successfully used LIO-SAM with Velodyne and RSLidar with the same IMU previously, but we are not able to make it work with the Ouster.

Was the problem solved? How?

Thank you for your help.

Cristian-wp, 2022-09-13:
Same problem, but with an Ouster OS0-64.

IvanVN, 2022-09-14:
> Same problem, but with an Ouster OS0-64.

FYI, in our case it resulted a problem with the Ouster driver [not providing proper ROS timestamp](https://github.com/ouster-lidar/ouster_example/pull/214) if not synchronized by PTP. We solved it applying the proposed patch to the driver, but apparently [this functionality has been already added recently](https://github.com/ouster-lidar/ouster_example/issues/74) to the latest release (we have not tested it yet).

tokoro10g, 2023-09-01:
For someone who run into this problem and found this issue, here are the solutions as of today:
- Launch driver.launch in the ouster-ros package with `timestamp_mode:=TIME_FROM_ROS_TIME` instead of `TIME_FROM_PTP_1588`.
- If you would like to use PTP and the timestamp of the data is about 37 sec behind the system clock, set `ptp_utc_tai_offset` to 0 for driver.launch in the ouster-ros package.