TixiaoShan/LIO-SAM #191: [lio_sam_mapOptmization-5] process has died

jjbecomespheh, 2021-01-15:
Hi Ti Xiao Shan,

Im sorry to trouble you, but I have followed the step for installation but I encountered mapOptimization crash and Im not sure how to fix it.

[093B3F78-EC9A-4A3D-A375-8BA4AC8ABD49](https://user-images.githubusercontent.com/50895766/104686435-28d89600-5738-11eb-8b2c-583ec1d770a0.jpeg)
[image]
[image]

jjbecomespheh, 2021-01-15:
I have installed 4.0.2 GTSAM according to the Readme, I also tried 4.0.3 GTSAM but it did not work as well.

jjbecomespheh, 2021-01-20:
Okay, I have solved the problem.

Initially, I upzip and move the gtsam folder into my src folder of my workspace. Which I follow the step
mkdir build && cd build
cmake -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF ..
sudo make install -j8
I also compile LIO-SAM afterwards.
and I encounter mapoptimization crashing, which is due to gtsam as mentioned by @TixiaoShan in the Readme.

So I rm gtsam from my workspace, put it inside my root folder instead and run
mkdir build && cd build
cmake -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF ..
sudo make install -j8
at there.
The problem was resolved.

closed 2021-01-20 by jjbecomespheh

alwynmathew, 2023-04-20:
@jjbecomespheh I installed GTSAM 4.0 with the ppa given in the README and was able to compile the package successfully following https://github.com/TixiaoShan/LIO-SAM/issues/206 in ROS Noetic (Ubuntu 20.04) but I still get `[lio_sam_mapOptmization-4] process has died [pid 174714, exit code -6` when I launch the package. Were you using ROS noetic here?

syedjameel, 2024-08-30:
It is because of the GTSAM lib, 
I followed the following env configuration and it works fine (for ROS noetic): 
https://cheeson.xyz/2023/10/09/LIO-SAM/

sahith-chada, 2024-09-12:
I followed the steps @syedjameel  mentioned to install GTSAM lib, but even then I got the same error ([lio_sam_mapOptmization-5] process has died [pid 1146996, exit code -6) when I launched the package and ran the rosbag.
I am using ROS Noetic. Is there any fix?

syedjameel, 2024-09-12:
@sahith-chada, make sure to uninstall other versions of gtsam libs before following that env configuration.