TixiaoShan/LIO-SAM #197: CMake Error at /opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake:113

Rotoslider, 2021-02-01:
i have a Jetson AGX running Jetpack 4.5.
I installed ROS melodic 
i ran the dependencies you listed and your install instructions
 When I run catkin_make is get the error:
```
CMake Error at /opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake:113
(message):
  Project 'cv_bridge' specifies '/usr/include/opencv' as an include dir,
  which is not found.  It does neither exist as an absolute directory nor in
  '${{prefix}}//usr/include/opencv' 
```

I had to edit the cv_bridgeConfig.cmake file line 96 to:
`set(_include_dirs "include;/usr/include;/usr/include/opencv4")`
Then I had to edit your utility.h line 17 to:
`#include <opencv2/opencv.hpp>`
this allowed it to compile

closed 2021-02-18 by stale[bot]

HappySamuel, 2021-05-17:
Hi @Rotoslider 

Thanks for sharing the solution. It helps me to compile LIO-SAM in Jetson AGX.

Best,
Samuel

qtw1998, 2021-08-07:
thx, bro I get it!! thx again!