TixiaoShan/LIO-SAM #202: Having issues with my own data.

Rotoslider, 2021-02-09:
This is not a error in the code its a lack of understanding by me. I did not know where else to post this so please forgive if I did in the wrong place.  I am new to ROS and Linux. So please bear with me. 
Setup is: Jetson AGX, Ubuntu 18, ROS Melodic.
Velodyne VLP-16 with Garmin antenna and a XSENS MTI-7 Development Kit 9 axis IMU with GPS and antenna.
 I installed LIO_SAM and can read and play back your sample bag files. I can see data from my Velodyne VLP-16 in rviz using" `roslaunch velodyne_pointcloud VLP16_points.launch calibration:=/home/lbackpack/Downloads/VLP-16.yaml`" and Then I launch LIO_SAM: `roslaunch lio_sam run.launch`
If I select Fixed frame in rviz as velodyne and under point cloud select /velodyne_points as the topic I can see the point cloud live.
I obviously am doing something wrong but searching for more than a week I cannot figure out the proper way to set this up. My IMU data is showing up in the sample code that comes with the MTi IMU. Its topic is imu_link. 
I think I need to edit the params.yaml file Topics section to the topic names velodyne_points and imu_link but it did not change anything. 
Could you offer some suggestions or links to where I can learn more about setting this up to work with LIO_SAM.

thanks for your time and software

Rotoslider, 2021-02-10:
I went through all the rostopics the IMU publishes and I actually needed imu/data instead of imu/link. I now have to sort out the Extrinsics. 
My IMU outputs:
```
rosmsg show sensor_msgs/Imu

std_msgs/Header header
  uint32 seq
  time stamp
  string frame_id
geometry_msgs/Quaternion orientation
  float64 x
  float64 y
  float64 z
  float64 w
float64[9] orientation_covariance
geometry_msgs/Vector3 angular_velocity
  float64 x
  float64 y
  float64 z
float64[9] angular_velocity_covariance
geometry_msgs/Vector3 linear_acceleration
  float64 x
  float64 y
  float64 z
float64[9] linear_acceleration_covariance
```
Will the above format work with LIO_SAM?
Currently I can see the point cloud and the blue line for the IMU but it jumps wildly and then stops in under a minute. I assume its either the format or the position of the IMU.
The IMU is X forward Y left and Z up
The Velodyne is setup like your image in the Readme. I have some confusion in that the Velodyne manual shows Y forward not X if its setup like how you picture it. 
Any suggestions?
thank you in advance

Rotoslider, 2021-02-13:
Seems like most issues here of late by everyone fall on deaf ears. But I will try anyway.
I used lidar_imu_calib to get the extrinsic numbers and enter them into the params.yaml
```
extrinsicRot: [0.928163, 0.357224, 0.104424,
                  0.263263, -0.828503, 0.494242,
                  0.262319, -0.431246, -0.86303]
  extrinsicRPY: [0.928163, 0.357224, 0.104424,
                  0.263263, -0.828503, 0.494242,
                  0.262319, -0.431246, -0.86303]
```
I uncomment the debug lines in "imuHandler()" of "imageProjection.cpp" so I could see the IMU data from LIO_SAM
```
 IMU acc: 
x: 1.0349, y: 4.56354, z: -8.62061
IMU gyro: 
x: -0.00505129, y: 0.00119266, z: -0.00181667
IMU roll pitch yaw: 
roll: -2.64178, pitch: -0.263641, yaw: 2.48605
```
This data is very different from my rostopic imu/data
```
header: 
  seq: 173683
  stamp: 
    secs: 1613187558
    nsecs: 764913859
  frame_id: "imu_link"
orientation: 
  x: 0.00228874385357
  y: 0.0174813140184
  z: 0.895720899105
  w: 0.44426715374
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.00156741612591
  y: -0.00206008600071
  z: 0.0018866736209
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: -0.112504370511
  y: 0.323832839727
  z: 9.80090236664
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```
Any ideas on why its so different and what I can do to fix it?
Velodyne VLP16 and Xsens MTi7 IMU

thanks in advance

Rotoslider, 2021-02-16:
I am having a issue with the gyro numbers. They bounce around wildly even when the IMU is still. And they do not seem to change in any reasonable form when the IMU is moved
I ran the IMU debug and using your video all work correctly as each axis is tilted except the gyro numbers. 
Z positive is correct
Roll Pitch and Yaw are correct
Acceleration is correct
Here is a sample from the debug frame.. The IMU has been sitting idle for more than 20 minutes. I started LIO_SAM and ran it for 1 minute. Then I shut it off. I scolled up through the data and grabbed this subset of data. You can see wide variations in gyo numbers. The IMU has not been touched for 20 minutes and nothing around it has moved to no magnetic changes. 
```
IMU acc: 
x: -0.0949261, y: -0.0744496, z: 9.80095
IMU gyro: 
x: 0.000661359, y: 0.00190436, z: 0.00187009
IMU roll pitch yaw: 
roll: -0.00745948, pitch: 0.00824724, yaw: -2.7853

IMU acc: 
x: -0.11061, y: -0.0845047, z: 9.8034
IMU gyro: 
x: -1.49012e-06, y: -0.000368431, z: 0.0026593
IMU roll pitch yaw: 
roll: -0.00747069, pitch: 0.00826267, yaw: -2.78526

IMU acc: 
x: -0.114606, y: -0.0620739, z: 9.79211
IMU gyro: 
x: 8.58679e-05, y: -0.00149012, z: 0.00167433
IMU roll pitch yaw: 
roll: -0.00747813, pitch: 0.00822297, yaw: -2.78524

IMU acc: 
x: -0.109925, y: -0.0817162, z: 9.81696
IMU gyro: 
x: -0.00123146, y: 0.000181217, z: 0.00113293
IMU roll pitch yaw: 
roll: -0.00749143, pitch: 0.00821132, yaw: -2.78524

```
Any suggestion on what I should look for or change to get the gyro numbers stable?

Rotoslider, 2021-02-17:
Attached is a pic of the point cloud. It build for a few seconds then rotates and build then rotates. Drifting is not so much of a problem as before. 
The Gyro output in Lio is still wrong but it works correctly in the Xsens MTI software 
[image]
@TixiaoShan Could you offer any help here if you have time?

Rotoslider, 2021-02-21:
I can not seem to get my head wrapped around this for some reason.
For starters I am confused about the matrix and what each value represents.:
```
extrinsicRot: [-1, 0, 0,
                  0, 1, 0,
                  0, 0, -1]
```
Based on how the params.yaml is filled out for the default:
I assume first row is linear acceleration for X, Y, Z
The second row  is angular velocity for X, Y, Z
The third row is IMU orientation but the code shows W, X, Y, Z. But there is only 3 spots

 under the Prepare IMU data in the readme it states:
    _

> we need to set the readings of x-z acceleration and gyro negative to transform the IMU data in the lidar frame, which is indicated by "extrinsicRot" in "params.yaml."
>     The transformation of attitude readings is slightly different. We rotate the attitude measurements by -90 degrees around "lidar-z" axis and get the corresponding roll, pitch, and yaw readings in the lidar frame. This transformation is indicated by "extrinsicRPY" in "params.yaml."

_
Since i am confused on line 1 I see -1 means X accel is neg and  Y and Z did not change
On line 2 Y velocity is positive. X and Z unchanged
On line 3 I have no idea unless W is not used and then its Z angular velocity is neg and X and Y are unchanged.

I never took Linear Algebra so please excuse my ignorance. I am trying to learn but I am not understanding yet. 
I have my IMU aligned X front Z up and Y left. My VLP 16 is pointing to the front so based on what was said that the velodyne driver uses the REP103 standard it is in the same orientation as my IMU 
I have looked at time stamps and both are the same between the IMU and Lidar
If I run the debug code and look at the numbers based on the debug video all are the same including the gyro now. Before I was not noticing it. But when I move it matches the video. it does change either positive or negative depending on the the axis of rotation.
From what I can see from Xsens MTi IMU the rot and rpy should be the same. They state it should all be 0,s unless NWU or NED. I have tried both NWu and NED matrix values from the and they made 0 difference. 
Actually I can put any value in and it makes no difference to the good. Some numbers make it worse but nothing makes it better.

Rotoslider, 2021-02-21:
I used the numbers from imu_utils to get the noise values. I ran it with the lidar running for 2 hours since the imu is on the bottom of the puck. 
```
imuAccNoise: 9.6604653964982700e-03
  imuGyrNoise: 1.0210079015364618e-03
  imuAccBiasN: 3.2754761326074500e-04
  imuGyrBiasN: 2.7028654899653045e-05
 
```
I used imu_calib #'s and it made things worse. I set them back to:
```
extrinsicRot: [1, 0, 0,
                  0, 1, 0,
                  0, 0, 1]
  extrinsicRPY: [1, 0, 0,
                  0, 1, 0,
                  0, 0, 1]

```
And all my direction changes in the debug window  match when using the debug video.
But if I play my bag file it still jumps crazy all over the place. 
The bag file works good in SC-Lego-Loam. But I guess that is not using the IMU.

So if everything matches on the IMU debug for positive and neg based on direction of rotation why is the point cloud still bouncing all over in a outdoor dataset that works in Lego-Loam?

px4Li, 2021-02-22:
> Seems like most issues here of late by everyone fall on deaf ears. But I will try anyway.
>
> I used lidar_imu_calib to get the extrinsic numbers and enter them into the params.yaml
>
> ```
>
> extrinsicRot: [...]

How did you calibrate imu with lidar? I usually calibrate in small and simple room.

HMellor, 2021-02-22:
> For starters I am confused about the matrix and what each value represents

`extrinsicRot` and `extrinsicRPY` are 3 dimensional [rotation matrices](https://en.wikipedia.org/wiki/Rotation_matrix). So the interpretation you describe is incorrect. `extrinsicRot` is for the inertial data (accelerometer and gyroscope) and `extrinsicRPY` is for the orientation. 

> I used the numbers from imu_utils to get the noise values. I ran it with the lidar running for 2 hours since the imu is on the bottom of the puck.

This is correct

> I used imu_calib #'s and it made things worse. I set them back to:
> ```
> extrinsicRot: [1, 0, 0,
>                   0, 1, 0,
>                   0, 0, 1]
>   extrinsicRPY: [1, 0, 0,
>                   0, 1, 0,
>                   0, 0, 1]
> ```

Based on what you said about the physical orientation of your equipment, `extrinsicRot` should be the [identity matrix](https://en.wikipedia.org/wiki/Identity_matrix) (like you have it in the above quote), because there is no rotation required. One thing to check that isn't in the debug video is that the z gyro increases when yaw increases

> The bag file works good in SC-Lego-Loam. But I guess that is not using the IMU.

As for why it works for Lego-LOAM but not LIO-SAM, you have answered your question in the above quote. Something about your setup is incorrect. Have you set up `extrinsixTrans` as well?

Side note: @TixiaoShan, the diagram for IMU axes has yaw+ in the wrong direction. It doesn't follow the right hand rule, like all the other exes do.
[image]

Rotoslider, 2021-02-22:
> How did you calibrate imu with lidar? I usually calibrate in small and simple room.

I used [lidar_imu_calib](https://github.com/chennuo0125-HIT/lidar_imu_calib). I ran a 2 minute bag file which included the points and imu data. I did it in a small room. I did one moving the unit and one with it stationary. I used the results as the extrinsics in LioSam and the numbers from when I moved the unit seem to work better. But just using 100,0101,001 works just as well.
The noise numbers were from [imu_utils](https://github.com/gaowenliang/imu_utils)

Did you run your calibration with it moving or stationary? How long did you run the bag record for?

Rotoslider, 2021-02-22:
> 
> 
> > For starters I am confused about the matrix and what each value represents
> 
> `extrinsicRot` and `extrinsicRPY` are 3 dimensional [rotation matrices](https://en.wikipedia.org/wiki/Rotation_matrix). So the interpretation you describe is incorrect. `extrinsicRot` is for the inertial data (accelerometer and gyroscope) and `extrinsicRPY` is for the orientation.
> 
> > I used the numbers from imu_utils to get the noise values. I ran it with the lidar running for 2 hours since the imu is on the bottom of the puck.
> 
> This is correct
> 
> > I used imu_calib #'s and it made things worse. I set them back to:
> > ```
> > extrinsicRot: [1, 0, 0,
> >                   0, 1, 0,
> >                   0, 0, 1]
> >   extrinsicRPY: [1, 0, 0,
> >                   0, 1, 0,
> >                   0, 0, 1]
> > ```
> 
> Based on what you said about the physical orientation of your equipment, `extrinsicRot` should be the [identity matrix](https://en.wikipedia.org/wiki/Identity_matrix) (like you have it in the above quote), because there is no rotation required. One thing to check that isn't in the debug video is that the z gyro increases when yaw increases
> 
> > The bag file works good in SC-Lego-Loam. But I guess that is not using the IMU.
> 
> As for why it works for Lego-LOAM but not LIO-SAM, you have answered your question in the above quote. Something about your setup is incorrect. Have you set up `extrinsicTrans` as well?
> 
> Side note: @TixiaoShan, the diagram for IMU axes has yaw+ in the wrong direction. It doesn't follow the right hand rule, like all the other axes do.
> [image]

Thanks for the wiki links. Thats what I been needing to find. I will need to read through it again to maybe understand it. 
I went back and ran the debug video routine again in LioSam. And check the Z gyro as you suggested. It worked correctly. I hooked up the IMU to a windows box and used the Xsens software to watch all the axis movements. I followed the debug video again and everything rotated in the right direction as in the video. 
I have tried it with my extrinsicTrans numbers at 0.0, 0.0, 0.0 and at what they are supposed to be for my setup. The IMU is mounted directly under the lidar puck. -0.012, 0.0, -0.074
The IMU outputs at 100hz. Not sure if that would cause it to go wonky or not. 

Seems my orientation is correct. Maybe my IMU is bad. Seems like the angular velocity values are pretty noisy. peaking about 0.2 degrees each side of zero at rest with the lidar running using the Xsens software. The values I used from imu_utils should have canceled that out though I would think. 

Any other ideas? I don't have another $600+ just to see if a different IMU would fix the problem.

px4Li, 2021-02-22:
@Rotoslider 

maybe you should grab one rosbag that has 6 dof movement (front ,back, left,right, up,down,R,P,Y). I didn't grab the data longer than 20s. I am not sure this is correct but it works for me. BTW the extrinsicROT and extrinsicRPY are not same result.

Rotoslider, 2021-02-23:
> @Rotoslider
> 
> maybe you should grab one rosbag that has 6 dof movement (front ,back, left,right, up,down,R,P,Y). I didn't grab the data longer than 20s. I am not sure this is correct but it works for me. BTW the extrinsicROT and extrinsicRPY are not same result.

You didn't mention if your setup was stationary or moving during your calibration routine. 
In my case the rpy and rot are the same I think.  Lidar_imu_calib spits out:
```
result euler angle(RPY) : -0.022443 -0.0102182 0.0340398
result extrinsic rotation matrix : 
  0.999369 -0.0337954 -0.0109733
 0.0340314   0.999177  0.0220805
  0.010218   -0.02244   0.999696

```
If you put in the result euler angle(RPY) values into this [calculator](http://www.andre-gaschler.com/rotationconverter/) the rotation matrix come up the same. Well really really close anyway. Its a touch different each time you run it.

But like I said I am most likely wrong.

closed 2021-03-21 by stale[bot]

xiaoguorui, 2021-05-22:
@Rotoslider I have the same issue with you. Have you solved the problem? Thanks for reply

Rotoslider, 2021-05-23:
> 
> 
> @Rotoslider I have the same issue with you. Have you solved the problem? Thanks for reply

Sadly no. I gave up on LIO-SAM. I am using StaticMapping now.  I am also waiting on a camera to try out LVI-SAM. Maybe I can get that one to work.