TixiaoShan/LIO-SAM #204: IMU to LiDAR transform bug

HMellor, 2021-02-19:
Hi @TixiaoShan, I think I have found a bug in the transformation from IMU frame to LiDAR frame. From what I have seen, LIO-SAM uses the IMU data in the following way:
1. [Rotate IMU data into LiDAR frame](https://github.com/TixiaoShan/LIO-SAM/blob/944cda4b8f7e1af02958dc1c71bdde7d46068443/include/utility.h#L247)
2. [Integrate IMU data](https://github.com/TixiaoShan/LIO-SAM/blob/944cda4b8f7e1af02958dc1c71bdde7d46068443/src/imuPreintegration.cpp#L473)

However, this does not account for IMU acceleration that is tangential to the LiDAR->IMU lever arm. This error is proportional to the the extrinsic translation between LiDAR and IMU (a.k.a. the lever arm) and the rotation speed of the LiDAR (_not the RPM_).

If we look at the following image showing a simplified 2D representation of the LiDAR+IMU at time 0 and time 1 we can see that the linear acceleration of the LiDAR and the integrated IMU data do not match
[image]
If we follow the same process as LIO-SAM:
1. Rotate the IMU frame around its own origin so that its frame matches the orientation of the LiDAR frame
    [image]
2. Integrate the new acceleration vector. We start at 0 velocity and have a time difference of 1s, giving us `[6, -2]` in the rotated IMU frame. However, it should be `[4, 0]` in the LiDAR frame
    [image]

This would be particularly bad if you had pure rotation around the LiDAR origin, because the IMU data would say that there is translation occurring even though the LiDAR is stationary in `xyz`

I tried to implement a fix but I wasn't able to just yet, so I am reporting the issue, in case you or someone else can do it faster than me.

TixiaoShan, 2021-02-21:
That's correct. My IMU and lidar distance in my sensor setup is usually around 0.1 meters so I didn't take this effect into account. If anyone could submit a PR that would be great!

manomitbal, 2021-02-27:
@TixiaoShan @HMellor I added a crude implementation to solve this exact same issue that I also noticed. I added the transformation code within the callbacks for IMU within imuPreintegration.cpp and imageProjection.cpp

HMellor, 2021-02-27:
@TixiaoShan @manomitbal, in the comments of that PR you mention that you are ignoring translation of the IMU. Ignoring the translation is reason that this issue exists, therefore you're PR doesn't solve this issue.

manomitbal, 2021-02-27:
@HMellor I think @TixiaoShan can back me up on this but the only translation input from the IMU are the accelerations which are quite privy to noise. Transforming the linear accelerations will end up causing delay and also add noise since it is not done from the sensor frame. I think imuPreintegration.cpp does a good job adjusting the biases and covers up for this noise using the incremental odom topic provided by mapOptmization.cpp added as a factor. I typically would prefer using a kalman filter from robot_localization package that automatically does this for you where you fuse the lidar odometry with IMU for the initial estimate. But the most critical thing is rotation.

HMellor, 2021-03-02:
I am fully aware of what happens when you displace linear accelerations. The gyroscope noise gets added to the accelerometer data by the angular velocity terms in the equations below. 

This is, of course, a trade-off. But noisy data that is in the correct direction is preferable to data with less noise that is in the wrong direction. This noise is proportional to the translation between the IMU and the LiDAR, so for shorter lever arms this noise addition is minimal.

Take the following configuration as an example, where:
* **I** is the IMU frame and **R** is the transform from the origin to **I**
* **L** is the LiDAR frame and **r** is the transform from the origin to **L**
* **ρ** is the transform from **I** to **L** in **I**'s frame

[image]

The following differentiation shows all contributions of the linear acceleration of **L**
[image]

Since the lever arm **ρ** is constant, we can remove its derivative terms
[image]

Finally, we turn it into a numerical equation that can be written as code
[image]

If we set **R** = 0 and rotate the **I** so that it is aligned with **L**, then the implementation of this equation is simplified, as seen in #212

TixiaoShan, 2021-03-03:
Sorry for the late reply due to my limited bandwidth.

I was checking my code and noticed that I used [line 288](https://github.com/TixiaoShan/LIO-SAM/blob/2d6e50d9432767477d664342ab065f7b580d68ea/src/imuPreintegration.cpp#L288) to transform lidar correction pose to IMU frame. Is transforming IMU acceleration data again necessary? Correct me if my logic is wrong:
1. Rotate IMU data to comply with ROS standard in IMU frame.
2. Integrate IMU data in IMU frame in imuPreintegration.cpp
3. Calculate lidar odometry in mapOptimization.cpp in lidar frame
4. Transform lidar odometry from lidar frame to IMU frame with [line 288](https://github.com/TixiaoShan/LIO-SAM/blob/2d6e50d9432767477d664342ab065f7b580d68ea/src/imuPreintegration.cpp#L288)
5. Optimize the graph in IMU frame.
[image]

HMellor, 2021-03-03:
My PR rotates and translates IMU data to LiDAR frame at the same time (i.e. in your `step 1`). 

This means that, for all subsequent steps, the LiDAR frame is the same as the IMU frame. This makes your `step 4` no longer required and the graph optimisation happens in the LiDAR frame.

I think that my solution is correct because it means the initial guess for `step 3` is in the LiDAR frame and has extra acceleration from LiDAR rotation removed (i.e. `[4, 0]` instead of `[6, -2]` from the toy example)

Do you agree?

shibowing, 2021-03-16:
Hi @HMellor,  does this help to solve the oscillation issue when the robot is stationary?  I noticed that 200hz imuPreintegration odometry sometimes is very jerk even though the imu and lidar are synced.

HMellor, 2021-03-16:
Hi @shibowing, I haven't seen that issue in a while but you're welcome to try the fix using [my branch](https://github.com/HMellor/LIO-SAM/tree/imu-displace-fix) (it's not yet merged into `master` here). I'll also add a comment on #171 explaining why I think that oscillation occurred.

shibowing, 2021-03-23:
Thank you very much for your help! I will check this.

Saki-Chen, 2021-04-16:
I use imu data collected by rt3000.
According to the manual of rt3000, extrinsicRot should be same as extrinsicRPY. But it only works when extrinsicRPY is set to identity. When I set extrinsicRPY to the value of extrinsicRot, it jumps up and down.
[GPS_Manual.pdf](https://github.com/TixiaoShan/LIO-SAM/files/6327479/GPS_Manual.pdf)

Saki-Chen, 2021-04-17:
> I use imu data collected by rt3000.
> According to the manual of rt3000, extrinsicRot should be same as extrinsicRPY. But it only works when extrinsicRPY is set to identity. When I set extrinsicRPY to [...]

I finally solve it by converting the orientation of imu from NED to ENU.

Saki-Chen, 2021-04-24:
In my practice with imu data，the initial pose of imu is set in navigation frame with gravity pointing down and there is no need to rotate direction of imu measurements. And lidar pose is calculated as
                                         T_navigation_lidar = T_navigation_imu * T_imu_lidar
Besides the velocity in the bodyframe of lidar is
                                         local_V_lidar = adjointMap(T_lidar_imu) * local_V_imu
where
                                        adjointMap([R|t]) = [          R        0 ]
                                                                        [ [t]x * R        R ]
                                                              [t]x = [   0  -tx   ty ]
                                                                        [  tx    0  -tx ]
                                                                        [-ty    tx    0 ]
is it more clear ?

closed 2021-06-03 by stale[bot]

FishInWave, 2021-12-05:
@HMellor 
Hi, I use your code to run KITTI 00. As you know, there are some blank time with no imu data in that sequence, e.g. 2011-10-03 12:56:10.884089710 —— 2011-10-03 12:56:12.483976376. During this period, your code will fly away but TiXiao Shan's will not. Do you know the reason as it seems your code is more mathematical?

HMellor, 2021-12-20:
Hi @FishInWave, I haven't worked on this in quite some time now, but I'll try and help if I can. Are these regions missing both accelerometer and gyroscope data, or just accelerometer?

If just accelerometer data is missing, then my code will produce non-zero data in **both** the transformed acceleration and transformed angular rate measurements, causing LIO-SAM see it as valid. However, the data in the transformed acceleration will be ~0 (assuming no extreme rotations and a relatively short lever arm), causing LIO-SAM to believe that the IMU is *falling* at ~9.81m/s. You could verify this by uncommenting the [IMU debug prints in `imageProjection.cpp`](https://github.com/TixiaoShan/LIO-SAM/blob/master/src/imageProjection.cpp#L152).

Please see the equation I posted in my [earlier comment](https://github.com/TixiaoShan/LIO-SAM/issues/204#issuecomment-788827042) to understand why this is the case:

> [image]

zyx-99, 2022-04-26:
> In my practice with imu data，the initial pose of imu is set in navigation frame with gravity pointing down and there is no need to rotate direction of imu measurements. And lidar pose is calculated as T_navigation_lidar = T_navigation_imu * T_imu_lidar Besides, the twist(compose angular rate and linear velocity in one vector) in the bodyframe of lidar is Twist_lidar = adjointMap(T_lidar_imu) * Twist_imu where adjointMap([R|t]) = [ R, 0; [t]x * R, R ] [t]x = [ 0, -tx, ty; tx, 0, -tx; -ty, tx, 0 ] for example, if we have device blow at the origin of navigation frame with x pointing forward and z pointing down at time zero, the rotation matrix R of initial T_navigation_imu can be calculated from rpy = [pi, 0, 0] [image] for the setup of device blow in lio-sam rpy = [0, pi, 0] [image] is it more clear ?

Hello, could I ask you a question. When you set the coordinate system to ENU, how the Lidar and IMU are placed and set extrinsicROT and extrinsicRPY. Thank you!!!!!

kekeliu-whu, 2022-11-17:
A better way is to transform lidar scan to IMU frame instead of reverse transformation. At the same time, time field is required to represent horizon angle.