TixiaoShan/LIO-SAM #206: catkin_make in ROS Noetic [Error]

HappySamuel, 2021-02-23:
I am trying to install the LIO-SAM in Ubuntu 20.04 and ROS Noetic. However i couldn't get it catkin_make.

Error 1:
GTSAM version: ppa gtsam (4.0.3)
Reason of using ppa GTSAM (4.0.3) is because failed to build GTSAM (4.0.2) from source. It failed at `cmake -DGTSAM_BUILD_WITH_MARCH_NATIVE=OFF ..`
[image]
Is there any solution to this error?

Error 2:
catkin_make ERROR
[image]

What's the opencv version needed for LIO-SAM?

Best,
Samuel

Rotoslider, 2021-02-23:
For me on Melodic with Opencv4.5 installed I had the change the line in utility.h to opencv2/opencv.hpp.

HappySamuel, 2021-02-24:
Hi @Rotoslider 

Thanks for the info. 

@TixiaoShan 
I have successful tried out installing LIO-SAM with ROS Noetic, and we need to do the following changes:
**Step 1:** Install GTSAM 4.0.3 binary
[image]
Reason of doing so, is because failed to install from source using the gtsam 4.0.2

**Step 2:** Configure the utility.h to use 
`#include <opencv2/opencv.hpp>` instead of `#include <opencv/cv.h>`

Step 3: Configure CMakeLists.txt to use 
`set(CMAKE_CXX_FLAGS "-std=c++14")` instead of `set(CMAKE_CXX_FLAGS "-std=c++11")`

With these 3 modifications, LIO-SAM is able to be installed on Ubuntu 20.04 and ROS Noetic. You may try it out.

Best,
Samuel

closed 2021-02-24 by HappySamuel

J-Rojas, 2021-06-15:
@HappySamuel  this was very helpful. @TixiaoShan  Can this be added to the documentation?

FrankBu0616, 2021-07-03:
@HappySamuel Thanks for the post! Supper helpful!

juliangaal, 2021-08-19:
I created [a fork](https://github.com/juliangaal/LIO-SAM) that compiles on Ubuntu 20.04 with ROS Noetic

willat343, 2022-03-25:
Even with @HappySamuel's changes, I still encountered a strange error:
```
error: ‘class std::unordered_map<unsigned int, std::vector<unsigned int> >’ has no member named ‘serialize’
```
which was resolved by moving `#include <opencv2/opencv.hpp>` after the `pcl` headers.

Also the stable `gtsam-release-4.0` at time of writing is now 4.0.3, so it was not necessary to install GTSAM from source.

This appears to be caused by a incompatibility between FLANN and OpenCV, whereby the `USE_UNORDERED_MAP` macro leeks from the OpenCV header and is picked by FLANN. Swapping the header order resolves this issue. See https://github.com/flann-lib/flann/issues/214 for more details.

TixiaoShan, 2022-04-11:
Thanks for providing information for **Noetic** compiling. To summarize [HappySamuel](https://github.com/HappySamuel) and [willat343](https://github.com/willat343)'s comments:
```
1. Install GTSAM following readme
2. Configure the utility.h to use **#include <opencv2/opencv.hpp>** instead of #include <opencv/cv.h>
3. Configure CMakeLists.txt to use **set(CMAKE_CXX_FLAGS "-std=c++14")** instead of `set(CMAKE_CXX_FLAGS "-std=c++11")`
4. Move #include <opencv2/opencv.hpp> after the pcl headers
```

leo-stan, 2022-05-09:
According to [this](https://stackoverflow.com/questions/42504592/flann-util-serialization-h-class-stdunordered-mapunsigned-int-stdvectorun) all that has to be done to fix the `flann` serialization error is to include `<pcl/kdtree/kdtree_flann.h>` before `<opencv2/opencv.hpp>`. I tested this change locally and it solved the issue.

Abhirup0209, 2022-10-04:
There are still some compilations issues with this now and ROS Noetic. Can anyone please help me?

X1yz, 2024-04-08:
> Hi All, I still have issues compiling in Noetic. Below is terminal feedback: [image] Can anyone help?

i have the same problem.Have you sovled it?

pospichal-lukas, 2024-07-03:
> > Hi All, I still have issues compiling in Noetic. Below is terminal feedback: [image] Can anyone help?
> 
> i have the same problem.Have you sovled it?

I ran catkin make again. The build succeeded the second time. Don't ask me why.

Kenneth-fuge, 2024-07-10:
把flann移到opencv前面
#include <pcl/kdtree/kdtree_flann.h>
// #include <opencv/cv.h>
#include <opencv2/opencv.hpp>