TixiaoShan/LIO-SAM #209: Ouster + External IMU inconsistent rendering and frequent jumps

hummblebee, 2021-03-01:
I have been trying to setup my Ouster OS1-16 (Gen1) with an XSENS MTi-680G and when I input the bag file data into LIO-SAM I get the following result with jumps and mirror like displacements:

[image]

Also after a few seconds the point cloud starts to shift and move while the data was recorded in a room and the sensors were stationary all the time!

I have read the README.md and un-synced timestamps are stated as a cause for these issues but I here's a sample output from the 3 first messages of topics for LiDAR, internal IMU and XSENS IMU in the bag and as you can see the timings are pretty close (almost 16 ms delay between the external IMU and the LiDAR) are these considered out of sync? Or am I missing something else?
```
LiDAR (t) timestamp:            1613148018414091000  -  February 12, 2021 4:40:18.414 PM
LiDAR (msg.header) timestamp:   1613148018409496784  -  February 12, 2021 4:40:18.409 PM
LiDAR (t) timestamp:            1613148018460989034  -  February 12, 2021 4:40:18.460 PM
LiDAR (msg.header) timestamp:   1613148018457116604  -  February 12, 2021 4:40:18.457 PM
LiDAR (t) timestamp:            1613148018515186248  -  February 12, 2021 4:40:18.515 PM
LiDAR (msg.header) timestamp:   1613148018511394739  -  February 12, 2021 4:40:18.511 PM
### rostopic delay output:
average delay: 0.002
	min: -0.006s max: 0.013s std dev: 0.00532s window: 12
average delay: 0.005
	min: -0.006s max: 0.037s std dev: 0.00780s window: 32
average delay: 0.004
	min: -0.006s max: 0.037s std dev: 0.00703s window: 52

(For comparison purposes)
Internal IMU (t) timestamp:             1613148018406862597  -  February 12, 2021 4:40:18.406 PM
Internal IMU (msg.header) timestamp:    1613148018406382799  -  February 12, 2021 4:40:18.406 PM
Internal IMU (t) timestamp:             1613148018412751736  -  February 12, 2021 4:40:18.412 PM
Internal IMU (msg.header) timestamp:    1613148018411708593  -  February 12, 2021 4:40:18.411 PM
Internal IMU (t) timestamp:             1613148018422281776  -  February 12, 2021 4:40:18.422 PM
Internal IMU (msg.header) timestamp:    1613148018421824217  -  February 12, 2021 4:40:18.421 PM
### rostopic delay output:
average delay: -0.000
	min: -0.009s max: 0.026s std dev: 0.00631s window: 99
average delay: -0.000
	min: -0.010s max: 0.026s std dev: 0.00628s window: 200
average delay: -0.000
	min: -0.014s max: 0.026s std dev: 0.00611s window: 300

XSENS IMU (t) timestamp:            1613148018392066417  -  February 12, 2021 4:40:18.392 PM
XSENS IMU (msg.header) timestamp:   1613148018390321087  -  February 12, 2021 4:40:18.390 PM
XSENS IMU (t) timestamp:            1613148018400302778  -  February 12, 2021 4:40:18.400 PM
XSENS IMU (msg.header) timestamp:   1613148018392688256  -  February 12, 2021 4:40:18.400 PM
XSENS IMU (t) timestamp:            1613148018400491818  -  February 12, 2021 4:40:18.400 PM
XSENS IMU (msg.header) timestamp:   1613148018394696183  -  February 12, 2021 4:40:18.394 PM
### rostopic delay output:
average delay: -0.002
	min: -0.009s max: 0.015s std dev: 0.00445s window: 114
average delay: 0.002
	min: -0.010s max: 0.055s std dev: 0.01255s window: 506
average delay: 0.003
	min: -0.010s max: 0.060s std dev: 0.01300s window: 914
```

Also one thing to note: I modified the Ouster driver from [here](https://github.com/ouster-lidar/ouster_example) to mimic the timestamp behavior of the Velodyne drivers where the `time` and `msg.header` are both timestamped in full epoch format unlike the original Ouster that stamps the `msg.header` in nanoseconds only!
But I don't think this modification is the root issue causing the behavior above since I tested with the original driver too and it wouldn't even render, saying (messages are too old and discarded in RViZ).

If anyone could shed some light on this, I'd be grateful.

hummblebee, 2021-03-01:
I also tried the same setup except this time setting the Ouster to timestamp using its internal clock (`TIME_FROM_INTERNAL_OSC`) but when I play the bag file I get an error in RViZ stating `Message removed because it is too old (frame=[os_sensor], stamp=[7410.940159360])`.

Here are the first 3 messages from this bag:
```
LiDAR (t) timestamp: 		1614616844904756332 - March 1, 2021 4:40:44.904 PM
LiDAR (msg.header) timestamp: 	7403340742010 	    - August 8, 2204 7:32:22.010 PM
LiDAR (t) timestamp: 		1614616844954839447 - March 1, 2021 4:40:44.954 PM
LiDAR (msg.header) timestamp: 	7403390709220 	    - August 9, 2204 9:25:09.220 AM
LiDAR (t) timestamp: 		1614616844999718884 - March 1, 2021 4:40:44.999 PM
LiDAR (msg.header) timestamp: 	7403440683890	    - August 9, 2204 11:18:03.890 PM

Internal IMU (t) timestamp: 		1614616844868188755 - March 1, 2021 4:40:44.868 PM
Internal IMU (msg.header) timestamp: 	7403370178520       - August 9, 2204 3:42:58.520 AM
Internal IMU (t) timestamp: 		1614616844878109171 - March 1, 2021 4:40:44.878 PM
Internal IMU (msg.header) timestamp: 	7403380178580       - August 9, 2204 6:29:38.580 AM
Internal IMU (t) timestamp: 		1614616844887993798 - March 1, 2021 4:40:44.887 PM
Internal IMU (msg.header) timestamp: 	7403390178810       - August 9, 2204 9:16:18.810 AM

XSENS IMU (t) timestamp: 		1614616844866481557 - March 1, 2021 4:40:44.866 PM
XSENS IMU (msg.header) timestamp: 	1614616844865712708 - March 1, 2021 4:40:44.865 PM
XSENS IMU (t) timestamp: 		1614616844866509771 - March 1, 2021 4:40:44.866 PM
XSENS IMU (msg.header) timestamp: 	1614616844865768865 - March 1, 2021 4:40:44.865 PM
XSENS IMU (t) timestamp: 		1614616844866521682 - March 1, 2021 4:40:44.866 PM
XSENS IMU (msg.header) timestamp: 	1614616844865800048 - March 1, 2021 4:40:44.865 PM
```

jiawenquan, 2021-03-02:
set imu 100hz or 200hz, [Adjust the parameters](https://github.com/TixiaoShan/LIO-SAM/pull/210#issue-582758427)

MaxikKot, 2021-03-02:
Hі, Hummblebee! І have the same problem with os1-16 sensor.

hummblebee, 2021-03-02:
@jiawenquan I have not tried with 100Hz but here's the result with IMU 20 Hz to make it on par with the LiDAR frequency:

[image]

Will try 100Hz later and report back.

hummblebee, 2021-03-02:
> Hі, Hummblebee! І have the same problem with os1-16 sensor.

Well it seems like Ouster issues with LIO-SAM are common, I'm still researching, if I find a fix I'll let you know ASAP.

jiawenquan, 2021-03-03:
> Hі, Hummblebee! І have the same problem with os1-16 sensor.

You may set too many unnecessary imu options, causing communication blockage

MaxikKot, 2021-03-04:
> 
> 
> I have been trying to integrate my Ouster OS1-16 (Gen1) with an XSENS MTi-680G and here's how I set them up:
> 
>     * Ouster OS1 is connected to the Ethernet port of laptop
> 
>     * MTi IMU sensor is connected to a USB port
> 
>     * I have set the Ouster to timestamp in PTP 1588 mode, and issued the command `sudo ptpd -i enp3s0 -M` in a Terminal to make the Ethernet chip broadcast timings (thanks to @px4Li for the tip [here](https://github.com/TixiaoShan/LIO-SAM/issues/207#issuecomment-787873217))
> 
> 
> Tested and it seems PTP timestamps are working correctly but when I input the bag file data into LIO-SAM I get the following result with jumps and displacements:
> 
> [image]
> 
> I have read the README.md and **un-synced timestamps** are stated as a cause for these issues but I here's a sample output from the 3 first messages of topics for LiDAR, internal IMU and XSENS IMU in the bag and as you can see the timings are pretty close, are these considered out of sync? Or am I missing something else?
> 
> ```
> LiDAR (t) timestamp: 		1614616639310819517 - March 1, 2021 4:37:19.310 PM
> LiDAR (msg.header) timestamp: 	1614616639254676480 - March 1, 2021 4:37:19.254 PM
> LiDAR (t) timestamp: 		1614616639374563536 - March 1, 2021 4:37:19.374 PM
> LiDAR (msg.header) timestamp: 	1614616639304678400 - March 1, 2021 4:37:19.304 PM
> LiDAR (t) timestamp: 		1614616639423463862 - March 1, 2021 4:37:19.423 PM
> LiDAR (msg.header) timestamp: 	1614616639354687744 - March 1, 2021 4:37:19.354 PM
> 
> 
> Internal IMU (t) timestamp: 		1614616639282313498 - March 1, 2021 4:37:19.282 PM
> Internal IMU (msg.header) timestamp: 	1614616639280500775 - March 1, 2021 4:37:19.280 PM
> Internal IMU (t) timestamp: 		1614616639291110001 - March 1, 2021 4:37:19.291 PM
> Internal IMU (msg.header) timestamp: 	1614616639290500695 - March 1, 2021 4:37:19.290 PM
> Internal IMU (t) timestamp: 		1614616639301079360 - March 1, 2021 4:37:19.301 PM
> Internal IMU (msg.header) timestamp: 	1614616639300518531 - March 1, 2021 4:37:19.300 PM
> 
> 
> XSENS IMU (t) timestamp: 		1614616639298836081 - March 1, 2021 4:37:19.298 PM
> XSENS IMU (msg.header) timestamp: 	1614616639298028272 - March 1, 2021 4:37:19.298 PM
> XSENS IMU (t) timestamp: 		1614616639299337757 - March 1, 2021 4:37:19.299 PM
> XSENS IMU (msg.header) timestamp: 	1614616639298052150 - March 1, 2021 4:37:19.298 PM
> XSENS IMU (t) timestamp: 		1614616639299362726 - March 1, 2021 4:37:19.299 PM
> XSENS IMU (msg.header) timestamp: 	1614616639298077029 - March 1, 2021 4:37:19.298 PM
> ```
> 
> rosbag info of the bag:
> 
> ```
> version:     2.0
> duration:    21.5s
> start:       Mar 01 2021 20:07:19.28 (1614616639.28)
> end:         Mar 01 2021 20:07:40.78 (1614616660.78)
> size:        327.3 MB
> messages:    17140
> compression: none [431/431 chunks]
> types:       sensor_msgs/Imu         [6a62c6daae103f4ff57a132d6f95cec2]
>              sensor_msgs/NavSatFix   [2d3a8cd499b9b4a0249fb98fd05cfa48]
>              sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181]
>              tf2_msgs/TFMessage      [94810edda583a504dfda3829e70d7eec]
> topics:      /gnss                     72 msgs @   3.9 Hz : sensor_msgs/NavSatFix  
>              /imu/data               7243 msgs @ 33.8 kHz : sensor_msgs/Imu        
>              /os_cloud_node/imu      2151 msgs @ 100.0 Hz : sensor_msgs/Imu        
>              /os_cloud_node/points    430 msgs @  21.1 Hz : sensor_msgs/PointCloud2
>              /tf                     7243 msgs @ 35.5 kHz : tf2_msgs/TFMessage     
>              /tf_static                 1 msg             : tf2_msgs/TFMessage
> ```
> 
> Here's LIO-SAM's output:
> 
> ```
> [ WARN] [1614621691.228977708, 1614616643.476415178]: Large velocity, reset IMU-preintegration!
> [ WARN] [1614621696.328799517, 1614616648.572775929]: Large velocity, reset IMU-preintegration!
> [ WARN] [1614621698.231852286, 1614616650.479392595]: Large velocity, reset IMU-preintegration!
[... 159 lines cut ...]
>                                0,    0,    0,    0.03, 0,    0,    0,     0,     0,     0,    0,    0,    0,    0,      0,
>                                0,    0,    0,    0,    0.03, 0,    0,     0,     0,     0,    0,    0,    0,    0,      0,
>                                0,    0,    0,    0,    0,    0.1,  0,     0,     0,     0,    0,    0,    0,    0,      0,
>                                0,    0,    0,    0,    0,    0,    0.25,  0,     0,     0,    0,    0,    0,    0,      0,
>                                0,    0,    0,    0,    0,    0,    0,     0.25,  0,     0,    0,    0,    0,    0,      0,
>                                0,    0,    0,    0,    0,    0,    0,     0,     0.04,  0,    0,    0,    0,    0,      0,
>                                0,    0,    0,    0,    0,    0,    0,     0,     0,     0.01, 0,    0,    0,    0,      0,
>                                0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0.01, 0,    0,    0,      0,
>                                0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0.5,  0,    0,      0,
>                                0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,    0.01, 0,      0,
>                                0,    0,    0,    0,    0,    0,    0,     0,     0,     0,    0,    0,    0,    0.01,   0,
[... cut ...]
I noticed that the rotational speed of your OS1-16 is 20 Hz. however, Mr. Tixiao Shan's description says that the file imageProjection.cpp "" deskewPoint () "function utilizes this relative time to obtain the transformation of this point relative to the beginning of the scan. When the lidar rotates at 10Hz, the timestamp of a point should vary between 0 and 0.1 seconds. If you are using other lidar sensors, you may need to change the name of this time channel and make sure that it is the relative time in a scan. "
Perhaps this is preventing the correct timing of the point cloud and imu synchronization? Or maybe you should try  OS-1 16 in 10 Hz mode?

bugdeveloper0, 2021-03-06:
@hummblebee   Hі, Hummblebee!  I am testing the lio-sam system with subt datase but I got a continuous warning for "Large velocity, reset IMU-preintegration!". And the trajectory result is very bad. Did you test lio-sam with subt dataset？

hummblebee, 2021-03-08:
@MaxikKot Following your suggestions, I tried recording with LiDAR in 10Hz and different IMU frequencies including 100Hz but the issue still persists!

I got this error message from [Google Cartographer ROS](https://github.com/cartographer-project/cartographer_ros) maybe it gives someone a clue:
```
Check failed: msg->linear_acceleration_covariance[0] != -1 (-1 vs. -1) Your IMU data claims to not contain linear acceleration measurements by setting linear_acceleration_covariance[0] to -1. Cartographer requires this data to work. See http://docs.ros.org/api/sensor_msgs/html/msg/Imu.html.
```

hummblebee, 2021-03-08:
> @hummblebee Hі, Hummblebee! I am testing the lio-sam system with subt datase but I got a continuous warning for "Large velocity, reset IMU-preintegration!". And the trajectory result is very bad. Did you test lio-sam with subt dataset？

Hi @bugdeveloper0 no I have not, I did try with a Velodyne dataset I downloaded from the internet and also a sample small dataset from Ouster's website and they both produced good results.

hummblebee, 2021-03-08:
I found this excerpt from XSENS knowledge base on integrating MTi sensors with Velodyne LiDARS:

> In contrast to the MTi-G-710, the MTi-680G does not yet offer a “true” 1 PPS output that comes straight from the internal GNSS receiver. Instead, by using the Interval Transition Measurement synchronization feature, the MTi-680G can be configured to generate its own 1 PPS signal that is synchronized with the 1 PPS signal of the internal receiver. This pulse will be synchronized with the internal GNSS 1 PPS pulse in terms of frequency, but not in terms of phase. This means that the 1 PPS output of the MTi-680G does not appear at the exact start of each UTC second. The timing of the pulse depends on the moment you power up the MTi-680G.

>  The MTi-680G does provide sub-second data in its NMEA messages, however some lidar brands do not copy the full UTC time information from the $GPGGA or $GPRMC packets: They often assume that the 1 PPS signal and its corresponding data packet coincide with the start of a UTC second, and therefore the sub-seconds field is assumed to be zero. This can cause a data timing error of up to 1 second.

> Further guidelines for the MTi-680G will be added as soon as it is able to output a "true" GNSS 1 PPS signal.

Need expert opinion on this, do you believe this is the reason I'm getting bad results?

MaxikKot, 2021-03-08:
Values in matrices  extrinsicRot: and extrinsicRPY correspond to the location of the imu relative to the ouster?

MaxikKot, 2021-03-08:
> I found this excerpt from XSENS knowledge base on integrating MTi sensors with Velodyne LiDARS:
>
>
>
>
> Need expert opinion on this, do you believe this is the reason I'm getting bad results?

As I understand from your previous messages, you did not connect MTi-G-710 to ouster via pps, but synchronized via ptp? if so, what is described cannot be the cause of bad results.

JJuhyeon-Kim, 2021-03-12:
> Hi @bugdeveloper0 no I have not, I did try with a Velodyne dataset I downloaded from the internet and also a sample small dataset from Ouster's website and they both produced good results.

Hi, @hummblebee

Do you think there is mapping problem because we use ptpd?
According to the issue https://github.com/TixiaoShan/LIO-SAM/issues/94, it says that the timestamp should be synchronized precisely.
Shouldn't we use GPS or hardware PTP to synchronize timestamp?

hummblebee, 2021-03-17:
> Values in matrices extrinsicRot: and extrinsicRPY correspond to the location of the imu relative to the ouster? Does the Horizon_SCAN value correspond to the horizontal resolution of the scanner at 10 Hz?

I'm not sure about the extrinsic matrices, I went ahead with identity matrices just to rule out any misalignments but I highly doubt that identity matrices cause jumps! I set the Ouster's LiDAR mode to `2048x10` so `Horizon_SCAN` is set to 2048 currently (but was 1024 when I first created this issue).

hummblebee, 2021-03-17:
> Hi, @hummblebee
>
> Do you think there is mapping problem because we use ptpd?
> According to the issue #94, it says that the timestamp should be synchronized precisely.
> Shouldn't we use GPS or hardware [...]

@JJuhyeon-Kim It may be, I can't think of anything else. If someone could upload a bag file of an Ouster and an external IMU synced properly I could analyze the timings and maybe come up with a solution.

hummblebee, 2021-03-17:
I managed to sync the timestamps and here's what I got:

[image]

The jumps are gone! YAY! But the projected data is kind of vibrating in-place! I did see #120 but that one is not distorted like my case, as if the vibration does not register a new cloud over the previous one, but this is the case in mine!

MaxikKot, 2021-03-18:
Try to record your bag outdoors rather than indoors. And  set the white noise parameters for your INS in params.yaml. imuAccNoise etc.

JJuhyeon-Kim, 2021-03-19:
Hi, @hummblebee ! 

First of all, thank you for always answering my questions kindly even though you must be very busy.
I think I solved the problem.

I saw in the ouster driver that ros timestamp is not accurate.
So I didn't consider the issue.
However, using a ros timestamp was effective for me. (I recommend you to check issue#https://github.com/TixiaoShan/LIO-SAM/issues/94, the commend of YushengWHU)
My imu sensor is the same as the author's, and imu sensor's timestamp is also ros timestamp.

When I used ros timestamp, the rostopic delay was much smaller than when I used ptpd.
[image]

If you want to use ros timestamp, then check this issue https://github.com/ouster-lidar/ouster_example/issues/74.

The figure below is what I obtained.
[image]

Thank you for your kindness.
I hope your research goes well!

hummblebee, 2021-04-10:
Hi @JJuhyeon-Kim 
Thanks for sharing your experience, and sorry for the late reply! I'm glad you got it working. Unfortunately I abandoned the project as it was consuming too much time and getting nowhere, but I'll leave this issue open as long as possible, if it is any help for others.

Good luck to you all.

closed 2021-06-03 by stale[bot]