TixiaoShan/LIO-SAM #215: Why not use the output of ICP as the loop constrain edge for iSAM?

FishInWave, 2021-03-09:
Hi @TixiaoShan ,I'm a little confused about the data in loopPoseQueue.
These are the relevant codes：
```cpp
        // Get pose transformation
        float x, y, z, roll, pitch, yaw;
        Eigen::Affine3f correctionLidarFrame;
        correctionLidarFrame = icp.getFinalTransformation();
        // transform from world origin to wrong pose
        Eigen::Affine3f tWrong = pclPointToAffine3f(copy_cloudKeyPoses6D->points[loopKeyCur]);
        // transform from world origin to corrected pose
        Eigen::Affine3f tCorrect = correctionLidarFrame * tWrong;// pre-multiplying -> successive rotation about a fixed frame
        pcl::getTranslationAndEulerAngles (tCorrect, x, y, z, roll, pitch, yaw);
        gtsam::Pose3 poseFrom = Pose3(Rot3::RzRyRx(roll, pitch, yaw), Point3(x, y, z));
        gtsam::Pose3 poseTo = pclPointTogtsamPose3(copy_cloudKeyPoses6D->points[loopKeyPre]);

        loopPoseQueue.push_back(poseFrom.between(poseTo));
```
Why not use the output of ICP, i.e. correctionLidarFrame, as the loop constrain edge for iSAM directly?
And suppose the world origin is W, the current pose is C, the history loop frame pose is H, then `correctionLidarFrame` = H2C, `tWrong` = W2C. Thus, `tCorrect` = H2C * W2C. I cannot understand what `tCorrect` means?

narutojxl, 2021-03-09:
`correctionLidarFrame` is the correction delta_transform, which is the global perturbation got from icp, not local perturation, because icp's source and target point clouds are already transformed into  the global frame. So we can view `correctionLidarFrame` as a transformation, which start  from history keyframe to current keyframe in the global frame perspective. Gtsam is hamilton, so multiplys left, not right.

FishInWave, 2021-03-12:
> `correctionLidarFrame` is the correction delta_transform, which is the global perturbation got from icp, not local perturation, because icp's source and target point clouds are already transformed [...]

@narutojxl  Can you explain it in more detail as I'm new to Gtsam? You said  "correctionLidarFrame can be viewd as a transformation, which start from history keyframe to current keyframe in the global frame perspective", does it mean that “correctionLidarFrame = H2C ” ? And does Gtsam require the between Factor in local frame or global frame?

narutojxl, 2021-03-15:
* If icp's source and target point clouds are all in their own's local frames, then you will get `H2C` transform, which transforms points from source frame into target frame, namely transforms points from current keyframe into history keyframe. Meanwhile, this transform is  current keyframe's pose with respect to history keyframe.  But in code, author already transformed them into global frame, so  `correctionLidarFrame` is in the global frame. It is a global perturbation of current keyframe, not local.
* Because gtsam's pose  is hamilton, with a global perturbation to corret current keyframe, we multiply it left on the old pose, not right.
* Gtsam's between factor between A pose and B pose is A respect to B , or B respect to A, which depends on the order of you call  this API. It is a local transform, not global.

FishInWave, 2021-03-15:
> * If icp's source and target point clouds are all in their own's local frames, then you will get `H2C` transform, which transforms points from source frame into target frame, namely transforms [...]

Thank you for your patient explanation. If the pose in the global frame is specified as `P'`, then `correctionLidarFrame = H'2C'`, and `tWrong = W'2C'`.Thus, is this equation that `tCorrect = correctionLidarFrame * tWrong = H'2C' * W'2C’ = H2C ` right as you said that a local transform is needed by GTSAM's between factor?

narutojxl, 2021-03-15:
Not true, `tCorrect` is corrected current keyframe's pose in global frame.

FishInWave, 2021-03-18:
> Not true, `tCorrect` is corrected current keyframe's pose in global frame.

Thanks. I finally understand what you said.
I'm confused about how to understand `tCorrect = correctionLidarFrame * tWrong`  mathematically. For example,What would happen on this equation if we exchange the source cloud and target cloud of ICP?

narutojxl, 2021-03-19:
* If we exchange source and target clouds of icp in code,  i guess maybe we will get history keyframe's global perturbation, i don't know how to derivate its meaning in mathematics, if we don't do a test to verify.
* If we do not firstly change source and target clouds into global, namely souce point clouds are in current keyframe, target point clouds are in history keyframe, then what's the meaning of icp's transformation, does it transform points from corrected current keyframe into history?  How to derivate its meaning in mathematics?  To be honest, I have no idea. 
* Hi dear knowledgeable author, @TixiaoShan, how did you derive?

HaisenbergPeng, 2021-04-03:
I hold the same doubts with FishInWave. Suppose source cloud and target cloud fully overlap with each other, then correctionLidarFrame is identity matrix, does it mean the current pose does not require correction based on the code? 
BTW, @narutojxl what do you mean by "BetweenFactor is in local frame"?

narutojxl, 2021-04-05:
If source and target fully overlap,  `correctionLidarFrame` should be `I`. But there always exist drift(small or big) in backend, this rarely happens. Gtsam's between factor between A pose and B pose is A respect to B , or B respect to A, which depends on the order of you call this API.

closed 2021-06-03 by stale[bot]